openwrt/packages #11190: Netdata: syslog error

borpin, 2020-01-31:
Maintainer: @BKPepe 
Environment: 
Model	GL-MT300N-V2
Architecture	MediaTek MT7628AN ver:1 eco:2
Firmware Version	OpenWrt 19.07.0 r10860-a3ffeb413b / LuCI openwrt-19.07 branch git-20.006.26738-35aa527

# Description:
This error has been present since upgrade to latest version.
An error is seen in the syslog attributed to the package netdata. This error appears every 2 seconds. the start of the error cannot be seen in the log.
```
Fri Jan 31 06:40:54 2020 daemon.err netdata[1853]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Fri Jan 31 06:40:56 2020 daemon.err netdata[1853]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Fri Jan 31 06:40:58 2020 daemon.err netdata[1853]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
```

Note that, on the first attempt to upgrade OpenWrt, resulted in a large number of errors from Netdata.

I subsequently removed Netdata, re applied the upgrade firmware, and reinstalled Netdata.

In the Netdata GUI it reports as this version

[image]

borpin, 2020-01-31:
I have fixed the symptoms, but I am not sure it should have been an issue in the first place.

Ref Issue https://github.com/openwrt/packages/issues/8968

TL;DR The solution is to disable the ipc plugin.

To help others:

Netdata works on default settings overwritten by settings in `/etc/netdata/netdata.conf`  You can see the settings in use by navigating to `http://IP_where_netdata_is_running:19999/netdata.conf`. 

SSH into the router and edit the `/etc/netdata/netdata.conf` file and add the following lines

```
[plugin:proc]
        ipc = no
```

You can add in the **nano** editor from the System -> Software menu on LuCI.

Reboot, and the log messages are gone.

HTH

neheb, 2020-02-23:
I take it this is solved?

borpin, 2020-02-23:
> I take it this is solved?

Ah yes, sorry should have closed.

closed 2020-02-23 by borpin

1715173329, 2021-04-13:
@borpin Could you please add this to https://github.com/openwrt/packages/blob/master/admin/netdata/files/netdata.conf?

I think it's useful.

BKPepe, 2021-04-13:
@1715173329 Fixed by this: https://github.com/openwrt/packages/pull/14053#issuecomment-813104543

1715173329, 2021-04-13:
> @1715173329 Fixed by this: [#14053 (comment)](https://github.com/openwrt/packages/pull/14053#issuecomment-813104543)

Oh, sorry for missing that. It's really great.