openwrt/packages #11195: runc: compile error

ElonH, 2020-01-31:
Maintainer: @G-M0N3Y-2503 
Environment: (arch: x86_64, openwrt branch: master)

Description:

```
go: cannot find main module, but found vendor.conf in /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc
	to create a module there, run:
	go mod init

```

<details>
<summary>Log Detail</summary>
<br />

```
# elonh @ X550JX in /mnt/A/BuildOpenWRT/openwrt on git:master o [23:14:00] 
$ make package/runc/{clean,compile} V=s -j1
make[1]: Entering directory '/mnt/A/BuildOpenWRT/openwrt'
make[2]: Entering directory '/mnt/A/BuildOpenWRT/openwrt/feeds/packages/utils/runc'
rm -rf /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80
rm -f /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/stamp/.runc_installed
rm -f /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/packages/runc.list /mnt/A/BuildOpenWRT/openwrt/staging_dir/host/packages/runc.list
make[2]: Leaving directory '/mnt/A/BuildOpenWRT/openwrt/feeds/packages/utils/runc'
time: package/feeds/packages/runc/clean#0.14#0.10#0.25
make[1]: Leaving directory '/mnt/A/BuildOpenWRT/openwrt'
make[1]: Entering directory '/mnt/A/BuildOpenWRT/openwrt'
make[2]: Entering directory '/mnt/A/BuildOpenWRT/openwrt/feeds/packages/lang/golang/golang'
make[2]: Leaving directory '/mnt/A/BuildOpenWRT/openwrt/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.37#0.18#0.45
make[2]: Entering directory '/mnt/A/BuildOpenWRT/openwrt/package/libs/toolchain'
echo "libc" >> /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/mnt/A/BuildOpenWRT/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.19#0.04#0.22
make[2]: Entering directory '/mnt/A/BuildOpenWRT/openwrt/feeds/packages/utils/runc'
touch /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.prepared_f7c4c31420199a1f31a4975c4b2508cc_6664517399ebbbc92a37c5bb081b5c53_check
. /mnt/A/BuildOpenWRT/openwrt/include/shell.sh; gzip -dc /mnt/A/BuildOpenWRT/openwrt/dl/runc-1.0.0-rc8+91-3e425f80.tar.gz | tar -C /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80
touch /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.prepared_f7c4c31420199a1f31a4975c4b2508cc_6664517399ebbbc92a37c5bb081b5c53
rm -f /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.configured_*
rm -f /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/stamp/.runc_installed
( cd /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80 ; mkdir -p /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/bin /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/cache /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/tmp ; files=$(find ./ -type d -a \( -path './.git' -o -path './.go_work' \) -prune -o \! -type d -print | sed 's|^\./||') ; if [ "1" != 1 ]; then code=$(echo "$files" | grep '\.\(c\|cc\|cpp\|go\|h\|hh\|hpp\|proto\|s\)$') ; testdata=$(echo "$files" | grep '\(^\|/\)testdata/') ; gomod=$(echo "$files" | grep '\(^\|/\)go\.\(mod\|sum\)$') ; for pattern in ; do extra=$(echo "$extra"; echo "$files" | grep "$pattern") ; done ; files=$(echo "$code"; echo "$testdata"; echo "$gomod"; echo "$extra") ; files=$(echo "$files" | grep -v '^[[:space:]]*$' | sort -u) ; fi ; IFS=$'\n' ; echo "Copying files from /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80 into /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc" ; for file in $files; do echo $file ; dest=/mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/$file ; mkdir -p $(dirname $dest) ; cp -fpR $file $dest ; done ; echo ; link_contents() { local src=$1 ; local dest=$2 ; local dirs dir base ; if [ -n "$(find $src -mindepth 1 -maxdepth 1 -name '*.go' \! -type d)" ]; then echo "$src is already a Go library" ; return 1 ; fi ; dirs=$(find $src -mindepth 1 -maxdepth 1 -type d) ; for dir in $dirs; do base=$(basename $dir) ; if [ -d $dest/$base ]; then case $dir in */usr/share/gocode/src/github.com/opencontainers/runc) echo "github.com/opencontainers/runc is already installed. Please check for circular dependencies." ;; *) link_contents $src/$base $dest/$base ;; esac ; else echo "...${src#/mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src}/$base" ; ln -sf $src/$base $dest/$base ; fi ; done ; } ; if [ "" != 1 ]; then if [ -d /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src ]; then echo "Symlinking directories from /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src into /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src" ; link_contents /mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src ; else echo "/mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src does not exist, skipping symlinks" ; fi ; else echo "Not building binaries, skipping symlinks" ; fi ; echo ; )
Copying files from /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80 into /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc
vendor.conf
notify_socket.go
man/runc-ps.8.md
man/runc-resume.8.md
man/runc-spec.8.md
man/runc-create.8.md
man/README.md
man/runc-list.8.md
man/runc-run.8.md
man/runc-pause.8.md
man/runc-events.8.md
man/runc-update.8.md
man/runc-delete.8.md
man/runc-kill.8.md
man/runc-restore.8.md
man/runc-start.8.md
man/runc.8.md
man/runc-exec.8.md
man/md2man-all.sh
man/runc-checkpoint.8.md
man/runc-state.8.md
start.go
delete.go
signals.go
[... 758 lines cut ...]
/mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.built
touch /mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80=runc-1.0.0-rc8+91-3e425f80 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/include -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80=runc-1.0.0-rc8+91-3e425f80 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/target-x86_64_musl/usr/include -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/mnt/A/BuildOpenWRT/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include " [... cut ...]
make[2]: *** [/mnt/A/BuildOpenWRT/openwrt/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.built] Error 2
make[2]: Leaving directory '/mnt/A/BuildOpenWRT/openwrt/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#3.24#1.32#4.42
package/Makefile:111: recipe for target 'package/feeds/packages/runc/compile' failed
make[1]: *** [package/feeds/packages/runc/compile] Error 2
make[1]: Leaving directory '/mnt/A/BuildOpenWRT/openwrt'
/mnt/A/BuildOpenWRT/openwrt/include/toplevel.mk:225: recipe for target 'package/runc/compile' failed
make: *** [package/runc/compile] Error 2

```

jefferyto, 2020-02-01:
I was able to compile runc with the x86-64 snapshot sdk with no issues. I suggest [cleaning your build environment](https://openwrt.org/docs/guide-developer/build-system/use-buildsystem#cleaning_up) and trying it again.

G-M0N3Y-2503, 2020-02-03:
I was able to successfully compile https://github.com/openwrt/packages/commit/561b535c368acc6ed703fde39c449ee195f113ec on https://github.com/openwrt/openwrt/commit/3b99d6763996d4cf8d59361f25046cc8086da332.

1715173329, 2020-02-03:
The problem is caused by a golang argument: `GO111MODULE`.
Disable it or **set to empty** by the following command:
```bash
go env -w "GO111MODULE=" && export GO111MODULE=""
```

jefferyto, 2020-02-03:
`GO111MODULE` should be [unexported](https://github.com/openwrt/packages/blob/623fc51eef1b4129f79964f0e56153c9f5eccd40/lang/golang/golang-values.mk#L72-L73) - having it set in your environment should not affect the build.

1715173329, 2020-02-03:
> `GO111MODULE` should be [unexported](https://github.com/openwrt/packages/blob/623fc51eef1b4129f79964f0e56153c9f5eccd40/lang/golang/golang-values.mk#L72-L73) - having it set in your environment should not affect the build.

Current environment does really will affect the build.

I think golang will read the config from current user.
When I simplely enable it via setting the variable (`export GO111MODULE="on"`), it does't affect the build.
But when I enable it via go command (`go env -w "GO111MODULE=on"`), and I get the same error as ElonH's.

jefferyto, 2020-02-03:
Oh I see what you mean, [`go env -w`](https://golang.org/doc/go1.13#go-env-w) is new for Go 1.13.

closed 2020-02-04 by ElonH

jefferyto, 2020-02-04:
Incorrectly reading from the user's environment configuration file should be fixed in #11228 (and #11229 for openwrt-19.07).

1715173329, 2020-02-04:
> Incorrectly reading from the user's environment configuration file should be fixed in #11228 (and #11229 for openwrt-19.07).

I‘ve tested it, but it doesn't seem to work.

jefferyto, 2020-02-05:
#11255 should fix this case (and for containerd/docker-ce); these packages are not using the default build process from golang-package.mk and so were not affected by the previous change.

1715173329, 2020-02-05:
All right. I've succeeded in compiling it even `GO111MODULE` was enable.
Thank you very much!