openwrt/packages #11200: modemmanager: Modem not in "connected" state, but can still ping internet

jonesmz, 2020-02-01:
Maintainer: @nickberry17 
Environment: ARM, ipq806x, 18.06.1

Description:
```
root@4f06a0:~# ifconfig wwan0 ; ping -c 1 -I wwan0 8.8.8.8
wwan0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:100.76.176.87  P-t-P:100.76.176.87  Mask:255.255.255.240
          inet6 addr: fe80::75d4:2509:5b0c:f4a1/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:199804 errors:0 dropped:0 overruns:0 frame:0
          TX packets:78003 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:26398566 (25.1 MiB)  TX bytes:40927123 (39.0 MiB)

PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=50 time=48.083 ms

--- 8.8.8.8 ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 48.083/48.083/48.083 ms

root@4f06a0:~# ubus call network.interface.wwan status
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 14852,
	"l3_device": "wwan0",
	"proto": "modemmanager",
	"updated": [
	        "addresses",
	        "routes"
	],
	"metric": 0,
	"dns_metric": 0,
	"delegation": true,
	"ipv4-address": [
	        {
		                "address": "100.76.176.87",
		                "mask": 28
	        }
	],
	"ipv6-address": [

	],
	"ipv6-prefix": [

	],
	"ipv6-prefix-assignment": [

	],
	"route": [
	        {
		                "target": "0.0.0.0",
		                "mask": 0,
		                "nexthop": "100.76.176.88",
		                "source": "100.76.176.87\/32"
	        }
	],
	"dns-server": [
	        "198.224.152.119",
	        "198.224.154.135"
	],
	"dns-search": [

	],
	"inactive": {
	        "ipv4-address": [

	        ],
	        "ipv6-address": [

	        ],
	        "route": [

	        ],
	        "dns-server": [

	        ],
	        "dns-search": [

	        ]
	},
	"data": {

	}
}

root@4f06a0:~# mmcli -m 1
  -----------------------------
  General  |         dbus path: /org/freedesktop/ModemManager1/Modem/1
           |         device id: 9368cc77934951340df56d704794a52ea08777e3
  -----------------------------
  Hardware |      manufacturer: Telit
           |             model: LM960A18
           | firmware revision: 32.00.123
           |    carrier config: default
           |      h/w revision: 1.0
           |         supported: gsm-umts, lte
           |           current: gsm-umts, lte
           |      equipment id: 357926100460067
  -----------------------------
  System   |            device: /sys/devices/platform/soc/8af8800.usb3/8a00000.dwc3/xhci-hcd.0.auto/usb2/2-1
           |           drivers: option, qmi_wwan
           |            plugin: Telit
           |      primary port: cdc-wdm0
           |             ports: ttyUSB0 (unknown), ttyUSB1 (unknown), ttyUSB2 (unknown),
           |                    cdc-wdm0 (qmi), wwan0 (net)
  -----------------------------
  Numbers  |               own:
  -----------------------------
  Status   |              lock: sim-pin2
           |    unlock retries: sim-pin (3), sim-pin2 (3), sim-puk (10), sim-puk2 (10)
           |             state: registered
           |       power state: on
           |       access tech: lte
           |    signal quality: 100% (recent)
  -----------------------------
  Modes    |         supported: allowed: 3g; preferred: none
           |                    allowed: 4g; preferred: none
           |                    allowed: 3g, 4g; preferred: 3g
           |                    allowed: 3g, 4g; preferred: 4g
           |           current: allowed: 3g, 4g; preferred: 4g
  -----------------------------
  Bands    |         supported: utran-1, utran-4, utran-6, utran-5, utran-8, utran-2,
           |                    eutran-2, eutran-4, eutran-5, eutran-13, eutran-48, eutran-66
           |           current: utran-1, utran-4, utran-5, utran-8, utran-2, eutran-2,
           |                    eutran-4, eutran-5, eutran-13, eutran-48, eutran-66
  -----------------------------
  IP       |         supported: ipv4, ipv6, ipv4v6
  -----------------------------
  3GPP     |              imei: 357926100460067
           |       operator id: 311480
           |      registration: home
  -----------------------------
  SIM      |         dbus path: /org/freedesktop/ModemManager1/SIM/0
  -----------------------------
  Bearer   |         dbus path: /org/freedesktop/ModemManager1/Bearer/1
```

I would have expected ModemManager to tear down the Bearer when it transitioned from the "Connected" state to the "Registered" state, and for Netifd to detect this and act on it by showing the interface as "down".

nickberry17, 2020-02-19:
No, it does not do that as of now.  You will likely need a hotplug script to monitor your interface and take the necessary action.  I have an older example over here: https://github.com/nickberry17/modem-manager-keepalive/blob/master/30-keepalive_modemmanager

feckert, 2023-09-13:
The issue is quit old and a lot of changes were made since the report  to [Modemmanger-Proto-Hanlder](https://github.com/openwrt/packages/commit/c15e94f6c5fbbaedd41fd74b930a2a205f80afc0) and [ModemManager](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/775/) it self to made improvements for the connection hanlding. I will close this. If this issue is still present with master or the upcomming openwrt-23.05 release, please open this again.

closed 2023-09-13 by feckert