openwrt/packages #11205: adblock package, dnscrypt-proxy support missing

makedir, 2020-02-01:
Why was dnscrypt-proxy removed as supported file output for the adblock package? Please add it back, so the output file is in the format per line:

host.com

Is there a way to declare the output format too somewhere manually? the default output server=/host.com/ isnt compatible with dnscrypt-proxy.

dibdot, 2020-02-02:
Just use dnsmasq to block ads (which is running in parallel anyway).

closed 2020-02-02 by dibdot

makedir, 2020-02-02:
No it does not, because you use it for dnscrypt-proxy with

option localservice '0'
list server '127.0.0.53#53'

option serversfile '/tmp/adb_list.overall'
is ignored then.

dibdot, 2020-02-02:
Then you can't use adblock for list generation unfortunately, I do not plan to bring back dnscrypt support.

makedir, 2020-02-02:
And why is that? I am speaking about dnscrypt-proxy v2 (https://github.com/DNSCrypt/dnscrypt-proxy) not v1, which is in development. A support would just mean to alter a few lines in the script.

dibdot, 2020-02-02:
Feel free to provide a PR for that ... it's much more than a "few lines" ... at least you have to make sure that all works with reporting, adblock query, suspend/resume etc.