openwrt/packages #11207: ddns-scripts_cloudflare.com-v4 script not working for wildcard entries

northorsouth, 2020-02-01:
Hey guys,

I just upgraded to firmware 19.07.1 and my wildcard ddns entry stopped working. I have a basic ddns entry that works (domain.tld) but the wildcard entry (formatted as *@domain.tld) gives this error in the logs: "sanitize on domain found characters outside allowed subset". I have also tried formatting the domain field as *.domain.tld, same result. And again, this worked until I upgraded the firmware.

psych0d0g, 2021-02-14:
Quick fix:
https://github.com/openwrt/packages/pull/14737

just add * to the DNS_CHARSET VAR in the file: /usr/lib/ddns/dynamic_dns_functions.sh

feckert, 2021-05-19:
Fixed with https://github.com/openwrt/packages/commit/93344aafa065ebfe63330e60f7d02ecdcf39c9b8

closed 2021-05-19 by feckert