openwrt/packages #11249: libs/rpcsvc-proto: host-pkg fails to compile

diizzyy, 2020-02-04:
Maintainer: @Andy2244 
Environment: Alpine Linux 3.11.3

Description:

```
make[3]: Entering directory '/home/diizzy/openwrt-20200203/openwrt/feeds/packages/libs/rpcsvc-proto'
CFLAGS="-O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include" CPPFLAGS="-I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/lib -L/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/lib -L/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/lib" make -j1 -C /home/diizzy/openwrt-20200203/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4
make[4]: Entering directory '/home/diizzy/openwrt-20200203/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4'
make  all-recursive
make[5]: Entering directory '/home/diizzy/openwrt-20200203/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4'
Making all in rpcgen
make[6]: Entering directory '/home/diizzy/openwrt-20200203/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4/rpcgen'
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_main.o -MD -MP -MF .deps/rpc_main.Tpo -c -o rpc_main.o rpc_main.c
mv -f .deps/rpc_main.Tpo .deps/rpc_main.Po
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_parse.o -MD -MP -MF .deps/rpc_parse.Tpo -c -o rpc_parse.o rpc_parse.c
mv -f .deps/rpc_parse.Tpo .deps/rpc_parse.Po
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_sample.o -MD -MP -MF .deps/rpc_sample.Tpo -c -o rpc_sample.o rpc_sample.c
mv -f .deps/rpc_sample.Tpo .deps/rpc_sample.Po
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_scan.o -MD -MP -MF .deps/rpc_scan.Tpo -c -o rpc_scan.o rpc_scan.c
mv -f .deps/rpc_scan.Tpo .deps/rpc_scan.Po
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_svcout.o -MD -MP -MF .deps/rpc_svcout.Tpo -c -o rpc_svcout.o rpc_svcout.c
mv -f .deps/rpc_svcout.Tpo .deps/rpc_svcout.Po
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_tblout.o -MD -MP -MF .deps/rpc_tblout.Tpo -c -o rpc_tblout.o rpc_tblout.c
mv -f .deps/rpc_tblout.Tpo .deps/rpc_tblout.Po
gcc -DHAVE_CONFIG_H -I. -I..   -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -MT rpc_util.o -MD -MP -MF .deps/rpc_util.Tpo -c -o rpc_util.o rpc_util.c
mv -f .deps/rpc_util.Tpo .deps/rpc_util.Po
gcc  -O2 -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/include -I/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/include  -L/home/diizzy/openwrt-20200203/openwrt/staging_dir/host/lib -L/home/diizzy/openwrt-20200203/openwrt/staging_dir/hostpkg/lib -L/home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/host/lib -o rpcgen rpc_clntout.o rpc_cout.o rpc_hout.o rpc_main.o rpc_parse.o rpc_sample.o rpc_scan.o rpc_svcout.o rpc_tblout.o rpc_util.o
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.o: in function `usage':
rpc_main.c:(.text+0x1a): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.c:(.text+0x3d): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.c:(.text+0x60): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.c:(.text+0x83): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.c:(.text+0xa6): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.o:rpc_main.c:(.text+0xc2): more undefined references to `libintl_gettext' follow
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.o: in function `main':
rpc_main.c:(.text.startup+0x43): undefined reference to `libintl_textdomain'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_main.c:(.text.startup+0x726): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_scan.o: in function `get_token.part.0':
rpc_scan.c:(.text+0x52e): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_scan.c:(.text+0x680): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_scan.c:(.text+0x710): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_scan.c:(.text+0x748): undefined reference to `libintl_gettext'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: rpc_scan.o:rpc_scan.c:(.text+0x8cf): more undefined references to `libintl_gettext' follow
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:370: rpcgen] Error 1
[... cut ...]
time: package/feeds/packages/rpcsvc-proto/host-compile#1.57#0.13#1.70
make[2]: *** [package/Makefile:111: package/feeds/packages/rpcsvc-proto/host/compile] Error 2
make[2]: Leaving directory '/home/diizzy/openwrt-20200203/openwrt'
make[1]: *** [package/Makefile:107: /home/diizzy/openwrt-20200203/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/diizzy/openwrt-20200203/openwrt'
make: *** [/home/diizzy/openwrt-20200203/openwrt/include/toplevel.mk:227: world] Error 2
```

neheb, 2020-02-05:
Does

HOST_BUILD_DEPENDS:=libiconv/host

work?

diizzyy, 2020-02-05:
Nope, neither does gettext-full/host

neheb, 2020-02-05:
Maybe

HOST_LDFLAGS += -lintl

micmac1, 2020-02-05:
I think this bug was reported already in Gentoo's bugzilla: [Link](https://bugs.gentoo.org/show_bug.cgi?id=637084)

A fix was proposed to upstream [here](https://github.com/thkukuk/rpcsvc-proto/pull/5) and merged. There seems to be a follow-up commit done by upstream: [link](https://github.com/thkukuk/rpcsvc-proto/commit/ccb01bf0b1cae9fc9ba2a2c1e7f3455926654743)

Upstream doesn't seem to do releases often, so maybe you want to manually backport these commits. I guess gettext-full/host would need to be added to HOST_BUILD_DEPENDS afterward.

closed 2020-02-09 by diizzyy

Andy2244, 2020-02-11:
sorry missed the last comment on `HOST_BUILD_DEPENDS:=gettext-full/host` so some targets fail, will setup a new PR to fix this.

PR #11288 (2020-02-09): rpcsvc-proto: switch to git release (2020-01-16)
Maintainer: me
Compile tested: arm, mips (master)
Run tested: arm/mvebu (master)

Description:
* switch to git release (2020-01-16)
* fixes #11249
commit 852eb51c (2020-02-09): rpcsvc-proto: switch to git release (2020-01-16)

* switch to git release (2020-01-16)
* fixes #11249