openwrt/packages #11263: pingcheck: Makefile does not create /etc/pingcheck/*.d directories

jonesmz, 2020-02-06:
Maintainer: @br101

Description: Readme at https://github.com/br101/pingcheck says that there are directories under /etc/pingcheck/ that have scripts called when certain events happen.

The makefile here https://github.com/openwrt/packages/blob/openwrt-19.07/net/pingcheck/Makefile does not create those directories.

br101, 2020-02-11:
I'm not convinced that it should create these empty directories... what's the opinion of OpenWRT maintainers?

jonesmz, 2020-02-11:
It's a discover-ability problem. Not having the directories available means an end-user won't know they exist. Access to online documentation isn't always easy to get. E.g. i've been out in the middle of nowhere where cellphones don't work needing to fix a router that doesn't talk to the internet before.

I'd also argue that providing a 4 line script file in each of the directories is a good idea as well.

```
#!/bin/sh
# ${INTERFACE} -- logical network interface (e.g. wan) which goes online or offline
# ${DEVICE} -- physical device (e.g. eth0) which goes online or offline
# ${GLOBAL} -- ONLINE or OFFLINE depending on wether device is online thru other interfaces

But that does increase the image footprint more than just having the empty directories would.

PR #11316 (2020-02-17): pingcheck: Update and add script directories
- Update to version with longer interface names.

- Add /etc/pingcheck/(on|off)line.d/ directories with an example
  script. Closes #11263

Maintainer: me
Compile tested: ar71xx, master
Run tested: ar71xx, master
commit 5c4a32ff (2020-02-17): pingcheck: Update and add script directories

- Update to version with longer interface names.

- Add /etc/pingcheck/(on|off)line.d/ directories with an example
  script. Closes #11263
PR #11976 (2020-04-27): pingcheck: Update and add script directories
- Update to version with longer interface names.

- Add /etc/pingcheck/(on|off)line.d/ directories with an example
  script. Closes #11263

Maintainer: me
Compile tested: ar71xx, Rambutan, OpenWrt 19.07.2, master
Run tested:  ar71xx, Rambutan, OpenWrt 19.07.2, master

Description:

Cherry picked from master
commit aa353e13 (2020-04-27): pingcheck: Update and add script directories

- Update to version with longer interface names.

- Add /etc/pingcheck/(on|off)line.d/ directories with an example
  script. Closes #11263
closed 2020-02-17 by neheb