openwrt/packages #11266: pingcheck: Pingcheck tells netifd to take down interfaces if their ping fails

jonesmz, 2020-02-06:
Maintainer: @br101

Instead, pingcheck should rely on the scripts described by the read me here https://github.com/br101/pingcheck to determine policy.

In my situation, i *don't* want pingcheck doing anything, I just want to know about the interface not being able to ping. But in order to do that I need to patch pingcheck to cut that behavior out.

br101, 2020-02-11:
Well it didn't know there are no scripts before... is this really a bug, or just sub-optimal log output?

jonesmz, 2020-02-11:
Bad log output.

Changing the log message to "Executing offline scripts, if any installed" or something would have prevented the confusion.

br101, 2020-02-12:
pingcheck could check if there are any scripts in the directory before outputting the message, but I don't think this is an issue important enough to spend time at the moment... patches welcome