openwrt/packages #11273: openvswitch (2.95) and vxlan broken in 19.07.01 but works in 18.06.7

metacoma, 2020-02-07:
Maintainer: @yousong 
**Environment:**
I reproduced this issue on openwrt x86 (in virtual machine) and on ramips-mt7621 (mikrotik rb750).

**Description:**
Vxlan doesn't work on openwrt 19.07.01 with openvswitch 2.11.1
But works in openwrt 18.06.7 with openvwitch 2.8.5

**Steps to reproduce:**
on ubuntu (in my case 18.04) create ovs bridge with vxlan port
 
```shell
ubuntu # ovs-vsctl add-br ovs-sw0
ubuntu # ovs-vsctl add-port vxlan0 -- set int vxlan0 type=vxlan options:remote_ip=192.168.66.2
ubuntu # ip a a 10.200.0.1/24 dev ovs-sw0
ubuntu # ovs-vsctl show
0667f7c4-2864-4cb6-9f7b-8c9622fce328
    Bridge "ovs-sw0"
        Port "vxlan0"
            Interface "vxlan0"
                type: vxlan
                options: {remote_ip="192.168.66.2"}
        Port "ovs-sw0"
            Interface "ovs-sw0"
                type: internal
    ovs_version: "2.9.5"
```
configure the same bridge on openwrt 19.07.01 with openvswitch 2.11.1
```shell
openwrt # ovs-vsctl add-br ovs-sw0
openwrt # ovs-vsctl add-port vxlan0 -- set int vxlan0 type=vxlan options:remote_ip=192.168.66.1
openwrt # ip a a 10.200.0.2/24 dev ovs-sw0
openwrt # ovs-vsctl show
cbc8bae4-c778-4157-b148-b6b9919b9cad
    Bridge "ovs-sw0"
        Port "vxlan0"
            Interface "vxlan0"
                type: vxlan
                options: {remote_ip="192.168.66.1"}
        Port "ovs-sw0"
            Interface "ovs-sw0"
                type: internal
    ovs_version: "2.11.1"
```
and then start ping, for example from ubuntu 
```
$ ping -c1 10.200.0.2
PING 10.200.0.2 (10.200.0.2) 56(84) bytes of data.
From 10.200.0.2 icmp_seq=1 Destination Host Unreachable

--- 10.200.0.2 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms
```
in tcpdump on openwrt i see VXLAN traffic, arp request from ubuntu and nothing else
```shell
root@OpenWrt:~# tcpdump -ni wg
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wg, link-type RAW (Raw IP), capture size 262144 bytes
00:53:37.398334 IP 192.168.66.1.34732 > 192.168.66.2.4789: VXLAN, flags [I] (0x08), vni 0
ARP, Request who-has 10.200.0.2 tell 10.200.0.1, length 28
```
Log messages in /var/log/openvswitch/ovs-vswitchd.log
```
root@OpenWrt:/tmp/log/openvswitch# tail -f *
==> ovs-vswitchd.log <==
2020-02-07T00:51:44.451Z|00024|ofproto_dpif|INFO|system@ovs-system: Datapath supports ct_orig_tuple6
2020-02-07T00:51:44.519Z|00025|bridge|INFO|bridge ovs-sw0: added interface ovs-sw0 on port 65534
2020-02-07T00:51:44.521Z|00026|bridge|INFO|bridge ovs-sw0: using datapath ID 00006604ab48cb4b
2020-02-07T00:51:44.522Z|00027|connmgr|INFO|ovs-sw0: added service controller "punix:/var/run/openvswitch/ovs-sw0.mgmt"
2020-02-07T00:52:10.208Z|00028|bridge|INFO|bridge ovs-sw0: added interface vxlan0 on port 1
2020-02-07T00:53:37.399Z|00001|ofproto_dpif_upcall(handler7)|INFO|received packet on unassociated datapath port 2
2020-02-07T00:53:38.427Z|00002|ofproto_dpif_upcall(handler7)|INFO|received packet on unassociated datapath port 2
2020-02-07T00:53:39.451Z|00001|ofproto_dpif_upcall(handler6)|INFO|received packet on unassociated datapath port 2
2020-02-07T00:53:40.475Z|00003|ofproto_dpif_upcall(handler7)|INFO|received packet on unassociated datapath port 2
2020-02-07T00:53:41.499Z|00004|ofproto_dpif_upcall(handler7)|INFO|received packet on unassociated datapath port 2
```

The same schema works without any issues on openwrt 18.06.7, openvswitch 2.8.5

Any ideas?
Thanks for reading! ;-)

metacoma, 2020-02-07:
Maybe this information can help
I tried to install packages from 18.06.7 to openwrt 19.07.1
```shell
openrwrt # cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.1'
DISTRIB_REVISION='r10911-c155900f66'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.1 r10911-c155900f66'
DISTRIB_TAINTS=''
```

changed package repos from 19.07.1 to 18.06.7
```shell
openwrt # cat /etc/opkg/distfeeds.conf
src/gz openwrt_core http://downloads.openwrt.org/releases/18.06.7/targets/ramips/mt7621/packages
src/gz openwrt_kmods http://downloads.openwrt.org/releases/19.07.1/targets/ramips/mt7621/kmods/4.14.167-1-2e88863ccdd594fb8e842df3c25842ee
src/gz openwrt_base http://downloads.openwrt.org/releases/18.06.7/packages/mipsel_24kc/base
src/gz openwrt_luci http://downloads.openwrt.org/releases/18.06.7/packages/mipsel_24kc/luci
src/gz openwrt_packages http://downloads.openwrt.org/releases/18.06.7/packages/mipsel_24kc/packages
src/gz openwrt_routing http://downloads.openwrt.org/releases/18.06.7/packages/mipsel_24kc/routing
src/gz openwrt_telephony http://downloads.openwrt.org/releases/18.06.7/packages/mipsel_24kc/telephony
```

and then install openvswitch packages
```shell
openwrt # opkg list-installed | grep -i openvswitch
kmod-openvswitch-intree - 4.14.167+2.11.1-2
kmod-openvswitch-vxlan - 4.14.167+2.11.1-2
openvswitch - 2.8.5-1
openvswitch-base - 2.8.5-1
openvswitch-ovn - 2.8.5-1
openvswitch-ovn-base - 2.8.5-1
openvswitch-ovn-controller - 2.8.5-1
openvswitch-ovn-controller-vtep - 2.8.5-1
openvswitch-ovn-detrace - 2.8.5-1
openvswitch-ovn-docker-overlay-driver - 2.8.5-1
openvswitch-ovn-docker-underlay-driver - 2.8.5-1
openvswitch-ovn-nbctl - 2.8.5-1
openvswitch-ovn-sbctl - 2.8.5-1
openvswitch-ovn-trace - 2.8.5-1
openvswitch-ovs-dpctl-top - 2.8.5-1
openvswitch-ovs-l3ping - 2.8.5-1
openvswitch-ovs-parse-backtrace - 2.8.5-1
openvswitch-ovs-pcap - 2.8.5-1
openvswitch-ovs-tcpdump - 2.8.5-1
openvswitch-ovs-tcpundump - 2.8.5-1
openvswitch-ovsdb-client - 2.8.5-1
openvswitch-vtep - 2.8.5-1
```

and created the same ovs bridge and got the same result, vxlan don't work.

From my point of view the problem in kmod-openvswitch/kmod-openvswitch-vxlan or in kernel, maybe I'm wrong because I'm a newbie in this area.

metacoma, 2020-02-07:
Well, I have good news:

I'm downgraded kmod-openvswitch and kmod-openvswitch-vxlan from 2.11.1 to 2.8.5 
```shell 
openwrt # opkg install --force-depends --force-downgrade ./kmod-openvswitch-vxlan_4.14.167\+2.8.5-1_mipsel_24kc.ipk
Downgrading kmod-openvswitch-vxlan on root from 4.14.167+2.11.1-2 to 4.14.167+2.8.5-1...
Configuring kmod-openvswitch-vxlan.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-openvswitch-vxlan:
 *      kernel (= 4.14.167-1-16b77c6fbb9d9192d39ab642ba67d11a)
```
```shell
openwrt # # opkg install --force-depends --force-downgrade ./kmod-openvswitch_4.14.167\+2.8.5-1_mipsel_24kc.ipk
Downgrading kmod-openvswitch on root from 4.14.167+2.11.1-2 to 4.14.167+2.8.5-1...
Configuring kmod-openvswitch.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-openvswitch:
 *      kernel (= 4.14.167-1-16b77c6fbb9d9192d39ab642ba67d11a)
```

```shell
openwrt # opkg list-installed | grep kmod-openvswitch
kmod-openvswitch - 4.14.167+2.8.5-1
kmod-openvswitch-vxlan - 4.14.167+2.8.5-1
```

and it worked!

```shell
openwrt # ping -c1 10.200.0.1
PING 10.200.0.1 (10.200.0.1): 56 data bytes
64 bytes from 10.200.0.1: seq=0 ttl=64 time=119.597 ms

--- 10.200.0.1 ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 119.597/119.597/119.597 ms
```

yousong, 2020-02-07:
> ```shell
> openwrt # opkg list-installed | grep -i openvswitch
> kmod-openvswitch-intree - 4.14.167+2.11.1-2
> kmod-openvswitch-vxlan - 4.14.167+2.11.1-2
> ```

"intree" flavour were built from openvswitch source code, while the other one is from linux kernel source code.  Mixing them together can produce undefined result.  That may be the problem.

metacoma, 2020-02-07:
> "intree" flavour were built from openvswitch source code, while the other one is from linux kernel source code. Mixing them together can produce undefined result. That may be the problem.
Thanks for reply, if i remember correctly, i am already tried with 'kmod-openvswitch' package without success, but i will try it again and report the result

yousong, 2020-02-07:
> Thanks for reply, if i remember correctly, i am already tried with 'kmod-openvswitch' package without success, but i will try it again and report the result

While testing please also be sure that the loaded kernel module are the one you want it to be, not the previously installed then loaded one.

yousong, 2020-02-21:
@metacoma  any updates?

yousong, 2020-04-18:
For too long this issue receives no updates, I close it for now.  Feel free to comment here to provide further updates.

closed 2020-04-18 by yousong