openwrt/packages #11278: nginx-util，The latest source code under the compiler error!!

zxlhhyccc, 2020-02-07:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

peter-stadler, 2020-02-07:
Ok, can you give me the target and how you are building it (on which system, sdk or openwrt checkout)?

zxlhhyccc, 2020-02-08:
> Ok, can you give me the target and how you are building it (on which system, sdk or openwrt checkout)?
> Edit: And the error of course (make package/nginx-util/compile -j1 V=s) …

Nginx-util builds on the master branch of openwrt. See the error message（make package/feeds/packages/nginx-util/compile -j1 V=s）:
[image]

peter-stadler, 2020-02-08:
I see, it should work with a new compiler at
`staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-g++`

You can build it by `make toolchain/install` (this can take a while). Maybe you have to remove it beforehand using `make toolchain/clean`. 

Or you can reset it:
`make distclean && cp feeds.conf.default feeds.conf; ./scripts/feeds update -a && ./scripts/feeds install -a && make menuconfig && make tools/install && make toolchain/install && make target/compile`

neheb, 2020-02-09:
maybe it should be brought back down to c++17 as GCC7 is still in the tree.

peter-stadler, 2020-02-09:
Ah, I see, thank you, I will do that :-)

peter-stadler, 2020-02-09:
Done it, I had also some other improvements/fixes and made a cumulative commit, should I split it better?

closed 2020-02-09 by neheb