openwrt/packages #11279: intl:install icu-full-data but don't work.

xmlf, 2020-02-08:
Maintainer: @mhei 
Environment: X86_64

Description:
I install php7-mod-intl ,but it's not work.
PHP version 7.4.1
intl ICU version 65.1
I need `Any-Latin;Latin-ASCII`, But icu only include:
```
Any-FCC
Any-FCD
Any-Hex
Any-Hex/C
Any-Hex/Java
Any-Hex/Perl
Any-Hex/Unicode
Any-Hex/XML
Any-Hex/XML10
Any-Lower
Any-Name
Any-NFC
Any-NFD
Any-NFKC
Any-NFKD
Any-Null
Any-Remove
Any-Title
Any-Upper
Hex-Any
Hex-Any/C
Hex-Any/Java
Hex-Any/Perl
Hex-Any/Unicode
Hex-Any/XML
Hex-Any/XML10
Name-Any
```
I also install package 'icu-full-data'. 
`opkg install icu-full-data_65.1-2_x86_64.ipk`
But  it still not work.
What should I do?  Please help me, thank you.

neheb, 2020-02-08:
ping @nxhack

nxhack, 2020-02-13:
When I searched on Google, I found a lot. Please give me some time.

nxhack, 2020-02-13:
Hi @xmlf 

Please let me know , is "CONFIG_PHP7_FULLICUDATA=y" set in build config?

xmlf, 2020-02-13:
@nxhack 
What should I do to see this value?

nxhack, 2020-02-13:
Perhaps you are using the package  (via opkg)?
Unfortunately, the packages published on OpenWrt site are not set to full-icu.
If you want to use full-icu, you need to build it yourself.

xmlf, 2020-02-13:
YES.
I use the package  via opkg.
Many programs have detected intl extensions, it is believed that the icu transliteration device can work normally. But because there is no icu data, there are many warnings or errors.

nxhack, 2020-02-13:
When we test the intl function, it looks like it works.

If you can, please give me a simple test script.

xmlf, 2020-02-13:
```
            $transliterator = Transliterator::create('Any-Latin;Latin-ASCII');
            $slug           = $transliterator->transliterate($slug);
```

nxhack, 2020-02-13:
```
root@OpenWrt:~# cat test.php
<?php
            $slug ='喜欢';
            $transliterator = Transliterator::create('Any-Latin;Latin-ASCII');
            $slug           = $transliterator->transliterate($slug);
	    echo $slug;
?>
root@OpenWrt:~# php-cli test.php 
xi huan
```

xmlf, 2020-02-13:
The package installed by opkg does not contain transliterator id `Any-Latin;Latin-ASCII`.
How do you make it work correctly.

nxhack, 2020-02-14:
Please show me icudt65l.dat size
```
ls -l /usr/share/icu/65.1/icudt65l.dat
```

xmlf, 2020-02-14:
`
ls -l /usr/share/icu/65.1/icudt65l.dat
-rw-r--r-- 1 root root 27977584 2020-02-05 05:51 ./usr/share/icu/65.1/icudt65l.dat`

nxhack, 2020-02-14:
hmm... Looks good

Are there any other logs when an error occurs?
```
logread
```

xmlf, 2020-02-14:
I want to confirm whether my installation method is correct first:
1. opkg install php7-mod-intl
2. opkg install icu-full-data_65.1-2_x86_64.ipk

nxhack, 2020-02-14:
OK Please wait until the environment is prepared.

nxhack, 2020-02-14:
```
oot@OpenWrt:~# opkg install php7-mod-intl
Installing php7-mod-intl (7.4.2-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/php7-mod-intl_7.4.2-2_x86_64.ipk
Installing libxml2 (2.9.10-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/libxml2_2.9.10-1_x86_64.ipk
Installing php7 (7.4.2-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/php7_7.4.2-2_x86_64.ipk
Installing icu (65.1-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/icu_65.1-2_x86_64.ipk
Configuring libxml2.
Configuring php7.
Configuring icu.
Configuring php7-mod-intl.
root@OpenWrt:~# opkg install icu-full-data
Installing icu-full-data (65.1-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/icu-full-data_65.1-2_x86_64.ipk
Configuring icu-full-data.
root@OpenWrt:~# opkg install php7-cli
Installing php7-cli (7.4.2-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/php7-cli_7.4.2-2_x86_64.ipk
Configuring php7-cli.
root@OpenWrt:~# cat test.php 
<?php
            $slug ='武汉加油！中国加油！';
            $transliterator = Transliterator::create('Any-Latin;Latin-ASCII');
            $slug           = $transliterator->transliterate($slug);
	    echo $slug;
?>

root@OpenWrt:~# php-cli test.php 
wu han jia you! zhong guo jia you!
```

xmlf, 2020-02-14:
I think I know what the problem is.
My opkg source from
http://bin.entware.net
I used Entware.
Entware does not have `icu-full-data`

xmlf, 2020-02-15:
@nxhack 
@nxhack
山川异域，风月同天。
Thank you very much.I will close this and submit issu in entware。Thank you again.

closed 2020-02-18 by neheb