openwrt/packages #11295: AdBlock: add exception for clients

LowSkillDeveloper, 2020-02-10:
Add the ability to turn off adblock for selected clients connected to the router.
For example, I want my desktop computer to show ads. And the rest of the computers and phones were left without ads.

dibdot, 2020-02-11:
Just configure a separate DNS server for that PC (a second local instance without adblock or an external one).

closed 2020-02-11 by dibdot

CAFxX, 2022-03-21:
I wanted to voice my +1 for this request. Beside the convenience and user-friendliness of actually having this function compared to the workaround, the workaround itself is not always applicable, e.g. when the client itself does not allow you to easily change the DNS server (I can count at least 2 appliances in my house that do not allow it out of the box) or when adblock itself is configured in "Force Local DNS" mode.