openwrt/packages #11298: <iptables> ! command not implemented for protocol matches

xbrico, 2020-02-10:
# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (mvbeu, CortexA9, OpenWRT 19.0.1)

Description:
When trying to use ! infront of a protocol to invert match when in iptmark command, iptables throws up an error. Code trying to use below (ipset works fine). Having to write code to specify 2 other protocols just gets messy and unweildy
```
iptmark -p !udp -m set --match-set latsens src,dst -j DSCP --set-dscp-class CS5 -m comment --comment "latency sensitive ipset" ## set dscp tag for Latency Sensitive (latsens) ipset
iptmark -p !udp -m set --match-set latsens src,dst -j RETURN
```

jow-, 2020-02-10:
This is an upstream iptables limitation.

xbrico, 2020-02-10:
Excellent closure notes and it works with iptables everywhere else, didnt have an issue on 19.0 - anywhere I can push this onto as there was no issue using this in last OpenWRT version?

jow-, 2020-02-10:
Sorry, your report subject and description were confusing, I thought you refer to iptables. No idea what `iptmark` is.

xbrico, 2020-02-10:
Right, apologies, my 1st time trying to post a bug so I am probably failing bad in my execution here. I am writing a DSCP marking script for IP(6)TABLES.

iptmark equals code:
```
iptmark(){
    $IPT -t mangle -A dscp_mark "$@"
}
ipt6mark() {
    $IPT6 -t mangle -A dscp_mark "$@"
}
```
whereas $IPT or $IPT6 = iptables or ip6 tables). So a basic example of where I use it is to wash all DSCP to CS0:
```
iptmark -j DSCP --set-dscp 0
ipt6mark -j DSCP --set-dscp 0
```

Where I am running into a problem is inversion match of a protocol as shown above:

```
iptmark -p !udp -m set --match-set latsens src,dst -j DSCP --set-dscp-class CS5 -m comment --comment "latency sensitive ipset" ## set dscp tag for Latency Sensitive (latsens) ipset
iptmark -p !udp -m set --match-set latsens src,dst -j RETURN
```

When I use -p !udp or -p ! udp as iptables would imply in its man file, I get an error return that !udp is an unknown protocol.

Just saw your update as I am writing this. I will try that as a fix and you can see if it matches up to what I just wrote! :)

Thanks for your time! :)

xbrico, 2020-02-10:
Looks like putting the inversion infront of -p as you suggested works so issue closed.

Thanks for your help.

closed 2020-02-10 by xbrico