openwrt/packages #11304: frr: libreadline7 missing

scaci, 2020-02-11:
Maintainer: @\lucize
Environment: (ARMv7 Processor rev 1 (v7l), Linksys WRT1200AC
, OpenWrt 19.07.1)

Description:

frr requires libreadline version 7, but openwrt 19.07.1 has version 8

workaround (seems to work): make 2 symlinks to library version 8:

libreadline.so.7 -> /usr/lib/libreadline.so.8
libreadline.so.7.0 -> /usr/lib/libreadline.so.8.0

```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2020-02-11:
Sounds like readline got bumped after frr. Meaning a rebuild is needed.

I think a PKG_RELEASE bump is all that's needed.

diizzyy, 2020-02-11:
Probably, I don't follow 19.X branch at all however

lucize, 2020-02-16:
frr I think is not on 19.x but it should have worked, please also update libyang

neheb, 2020-02-27:
I cannot find this in the repositories for 19.07. Closing.

closed 2020-02-27 by neheb