openwrt/packages #11325: ulogd: Can't run with musl

agomezxff, 2020-02-13:
Maintainer: @\neheb  (find it by checking history of the package Makefile)
Environment: OpenWrt SNAPSHOT r12152-1ab81bf02d 

Description:

Don't know if i'm missing something, even if it is here where report it, but can't run this daemon from a current snapshot with musl.

I managed to get it by commenting set_scheduler() call in ulogd.c, as it seems musl doesn't implement these scheduling functions.
Don't know how patches are written so have nothing to pull request

neheb, 2020-02-13:
Looking at the code, no idea why you need to comment it out.

agomezxff, 2020-02-16:
Oh, you are right. Last message is that warning but it's actually doing its work.
Thanks

closed 2020-02-16 by agomezxff