openwrt/packages #11336: Lang: add rust support

peterwillcn, 2020-02-14:
Add [rust-lang](https://www.rust-lang.org/) support

neheb, 2020-02-15:
Why?

klukonin, 2020-02-20:
I'll let myself to answer.

First of all, some packages such as Suricata require Rust to work properly.

https://github.com/openwrt/packages/pull/8659

champtar, 2020-02-22:
Feature requests are rarely effective, maybe try the forum or IRC
To respond to @neheb, rust-lang for embedded make more sense than golang or python, better performance and size, when the architecture is supported :(

neheb, 2020-02-22:
Sure. python is relatively huge.

I know go is large. No idea how big rust is.

champtar, 2020-02-22:
I don't have much experience in rust
Just did some tests and a stripped hello world is ~200k with the rust stdlib
There is no stable ABI yet in rust, so it seems that you have static binaries with everything in it except libc which is inked
no idea how hard it is to have separate libs, as for OpenWRT the solution around unstable ABI can be just rebuild the firmware

champtar, 2020-02-22:
Some people already played with rust on OpenWRT
https://medium.com/althea-mesh/cross-compiling-complex-rust-programs-for-openwrt-targets-80897fcf7648
https://github.com/althea-net/althea-packages/blob/master/althea/althea-rust-binaries/Makefile
https://github.com/japaric/rust-cross

champtar, 2020-02-22:
Not sure how slow and painfull it would be to rebuild the rust compiler & cargo like we do for golang.

jefferyto, 2020-02-22:
I've thought about adding Rust support for a while... My main concerns are:

* musl is in their [second tier](https://forge.rust-lang.org/release/platform-support.html#tier-2) of support, with most musl platforms only getting build tested with the standard library only. I don't know how stable that is, though I know almost nothing about Rust and maybe it is already very stable.

* I don't have a lot of free time to do this :joy: Building the Go build system took a while, even though I basically copied Debian's system. Plus I'm already busy enough supporting Python and Go, not sure I can add another language.

On the other hand, since I'm the one that brought Go packages into the feed, I feel like I need to make Rust packages possible to bring balance to the Force :joy:

diizzyy, 2020-03-15:
Not to mention that ~90%+ of all targets wont be able to run it anyway?

Grommish, 2021-11-25:
My PR is for including `rust-lang` into the build system, rather than putting `rustc` and `cargo` on-device.  Many of the arches aren't supported, or lack dependencies required (like LLVM support). It allows packages that utilize rust (like Suricata 6) be crossed for the target arch.  `rustc` and `cargo` on device would come later, if at all.

sndnvaps, 2022-09-21:
is there any progress on it.

oskarirauta, 2023-02-17:
I vouch for rust-lang support as well- unlikely that I will be ever writing anything in rust, just like I propably won't do with go either, and very likely I try to stay away from perl and python as well, while sticking with c/c++ on linux and swift on apple's family; these new languages seem to pop up every once in a while and it seems incredibly common that they become super popular in no time, personally, the biggest noticiable cave-at with go is the binary size (there's a system called upx by the way which reduces go binary sizes, I have not tried it though or know more than this about it)- although, I also try to stay away from cmake builds as much as possible, since go's and cmake's makefile structure is very confusing in my opinion..

Yet, I vouch for this, as like I said, very often these become super popular in no time, and for openwrt to keep up in the pace, we do seem to need rust-lang support; I've already crossed my paths with 2 packages that I thought could be interesting, until I noticed that they use something called rust (hadn't even heard about it by that time)- so it was a no go, popular containerization system podman, is about to abandon from cni networking and they are moving to competitive equillavent netavark; as I maintain podman and use it by myself, I am worried about this because, this netavark- is written in rust.

(NetAvark)[https://github.com/containers/netavark] doesn't bring any new bells and whistles for me personally, but I'd like to push updates to podman in the future as well, if it depends on netavark- openwrt drops out of that train unless we patch it to keep going with cni- which sounds bad idea update-wise..

While I am talking a lot about netavark here, it's just proof of concept that we do need rust-lang support for openwrt, as there was those 2 packages where other was zerotier's own dns management system or something like that, and something that I no longer remember- in a short time, rust seems to remind even me about it's existence, so it's very likely that it's that common that rust support is even required; and this comes from man who usually only vouches for c/c++/swift.

End-notes: why I don't like python? You all know cockpit project? A webui for managing some (not even that many) things mostly on centos linux? It has great podman management. Thing is just that python is very resource hungry, I had a VPS server where I hosted few containers for web serving purposes mainly, all things used to serve web content with php enabled, took a lot less resources that damn web ui(cockit) did- it doesn't sound very good when system stalls for webui, it doesn't even render graphics like graphical environment, such as kde or gnome might do- so when cpu is at 80% for cockpit and my vps get's shutdown by provider due to extensive cpu usage, there's something very badly wrong there. Later on I setup a custom openwrt environment and cpu usage is very low :) - and I also have webui if that's what I want to use for management, even with my own custom mini podman commander (just shows some specs of usage, containers and pods, and allows me to start/stop/restart pods/containers) - and system runs on very low resources.. leaving cpu utilization to where it's most needed there; content serving...

oskarirauta, 2023-03-27:
I have problems with rust. 

Such as:
```
  = note: /usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.2.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /usr/src/openwrt/staging_dir/hostpkg/cargo/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained/libc.a(sysconf.o): relocation R_X86_64_32S against `.rodata.values.0' can not be used when making a PIE object; recompile with -fPIE
          /usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.2.0/../../../../x86_64-openwrt-linux-musl/bin/ld: failed to set dynamic section sizes: bad value
          collect2: error: ld returned 1 exit status
          

error: could not compile `memchr` due to previous error
```

Should I re-build everything and set CONFIG_GCC_DEFAULT_PIE=y or what..?

stintel, 2023-04-01:
83785a7ce01613ed6408bed4d96bd5eea711bc01

closed 2023-04-01 by stintel

peterwillcn, 2026-10-30:
https://github.com/openwrt/openwrt/commit/429ccec2644e602b68fe5ebf83d9821c6aed191c