openwrt/packages #11351: hd-idle: appears not to work after reboot

craibuc, 2020-02-15:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: 

OS: `TurrisOS 4.0.5, Turris Omnia`

```bash
# opkg info hd-idle
Package: hd-idle
Version: 1.05-1.36
Depends: libc
Status: install user installed
Section: utils
Architecture: arm_cortex-a9_vfpv3
Size: 5328
Filename: hd-idle_1.05-1_arm_cortex-a9_vfpv3.ipk
Conffiles:
 /etc/config/hd-idle 4733fa40955720209aaba3fbd194b2b89cd845c36ada16f2c76f8efb138d7878
Description: hd-idle is a utility program for spinning-down external disks after a period of idle time.
Installed-Time: 1579549185
```

Description:

I've noticed after restarting the router that hd-idle doesn't seem to take effect.  If I forced (`hd-idle -i 300`), then the settings seem to apply.

There are no log entries (`dmesg | grep hd-idle`).

```bash
# ps ax | grep hd-idle
 3862 ?        S      0:04 /usr/bin/hd-idle -d -i 0 -a -i 300
10529 ?        S      0:01 hd-idle
10543 ?        S      0:02 hd-idle -i 300
21897 pts/0    R+     0:00 grep hd-idle
```
I'd like to enable logging, but `hd-idle -i 300 -l /var/log/hd-idle` doesn't seem to work.

If it makes a difference, I have a two-disk, RAID.

Reference: https://gitlab.labs.nic.cz/turris/turris-os-packages/issues/548

neheb, 2020-02-16:
TurrisOS 4 is based on 18.06 AFAIK. Maybe https://github.com/openwrt/packages/commit/024644cef3a3cac828251795b43c713fbec34150 needs to be backported...

ping @jefferyto

neheb, 2020-02-16:
Note that instead of fixing hd-idle, you could try using hdparm with its -B parameter to cause the hard drives to spin down.

BKPepe, 2020-02-16:
> TurrisOS 4 is based on 18.06 AFAIK. 

Yes, that's correct.

craibuc, 2020-02-16:
> Note that instead of fixing hd-idle, you could try using hdparm with its -B parameter to cause the hard drives to spin down.

I'm having issues w/ `hdparm`:

```bash
# hdparm -S 120 /dev/md0

/dev/md0:
setting standby to 120 (10 minutes)
HDIO_DRIVE_CMD(setidle) failed: Not a tty
```
Should I do this in `rc.local`, instead?

```bash
# cat /etc/rc.local
date >> ~/rc_local_executed

# Disable NCQ (fix RAID issue)
echo 1 > /sys/block/sdb/device/queue_depth
echo 1 > /sys/block/sdc/device/queue_depth
# /fix

# set disk sleep time to 10 minutes (10*60/5) ??
hdparm -S 120 /dev/sdb
hdparm -S 120 /dev/sdc

# start RAID array
# mdadm --assemble --scan
mdadm --assemble --force /dev/md0 /dev/sdb /dev/sdc

exit 0
```

neheb, 2020-02-16:
That...might be a problem with the SATA controller. Anyway, I meant -B not -S.

craibuc, 2020-02-16:
> That...might be a problem with the SATA controller. Anyway, I meant -B not -S.

```bash
# hdparm -B 120 /dev/md0

/dev/md0:
 setting Advanced Power Management level to 0x78 (120)
 HDIO_DRIVE_CMD failed: Not a tty
 HDIO_DRIVE_CMD(identify) failed: Not a tty
```

neheb, 2020-02-16:
Wait... Why are you running hdparm onto an MD device? It should be done on actual hard drives.

craibuc, 2020-02-16:
They are actually hard drives, in RAID 1 configuration.

neheb, 2020-02-17:
You're not listening.

/dev/md0 is a device created by mdadm. It is not a hard drive.

hdparm needs to be used on actual hard drives, not md devices.

jefferyto, 2020-02-18:
I doubt the init script in 18.06 is at fault (024644cef3a3cac828251795b43c713fbec34150 was not a big change).

I would be interested to see the content of `/etc/config/hd-idle` and what happens if you call the init script after booting (`/etc/init.d/hd-idle start`) instead of running hd-idle directly.

craibuc, 2020-02-18:
> I would be interested to see the content of `/etc/config/hd-idle` 
```
# cat /etc/config/hd-idle

config hd-idle
	option idle_time_unit 'minutes'
	option enabled '1'
	option idle_time_interval '10'
```
> and what happens if you call the init script after booting (`/etc/init.d/hd-idle start`) instead of running hd-idle directly.
to do

jefferyto, 2020-02-18:
From how the init script is written, I believe the [`disk` config option](https://openwrt.org/docs/guide-user/storage/hd-idle) is required. From the [relevant section](https://github.com/openwrt/packages/blob/openwrt-18.06/utils/hd-idle/files/hd-idle.init#L52-L59) of `hdidle_append()`:

```sh
	if [ "$numdisks" = "0" ]; then
		procd_open_instance
		procd_set_param command /usr/bin/hd-idle
		procd_append_param command -d -i 0
	fi
	procd_append_param command -a $disk
	procd_append_param command -i "$(compute_seconds $idle_time_interval $idle_time_unit)"
	numdisks=$(( numdisks + 1 ))
```

If there is no `disk` option, then the `-a` command line option will not have an argument (the device name or partial device name). I believe this argument is required if the `-a` command line option is present and hd-idle will exit if the argument is missing. You can try adding a `disk` config option to see if it works.

As it stands, it would be difficult to configure hd-idle with a default idle time for all hard drives (without specifying all of them in a `disk` option). It should be possible to amend the init script for this case, but it would probably require a change in the config file as well (I imagine the addition of a `default` section).

craibuc, 2020-02-19:
> I would be interested to see ... what happens if you call the init script after booting (`/etc/init.d/hd-idle start`) instead of running hd-idle directly.

After reboot, the service started and listed the correct number of files.  No need to manually run the init script.

jefferyto, 2020-02-19:
> After reboot, the service started and listed the correct number of files. No need to manually run the init script.

Did you change your config file?

craibuc, 2020-02-19:
> > After reboot, the service started and listed the correct number of files. No need to manually run the init script.
> 
> Did you change your config file?

Sorry, my mistake--this was in reference to `minidlna`.  

No changes to my config file.

> I would be interested to see ... what happens if you call the init script after booting (/etc/init.d/hd-idle start) instead of running hd-idle directly.

Trying this now.

```
# ps ax | grep hd-idle
 3923 ?        S      0:00 /usr/bin/hd-idle -d -i 0 -a -i 600
 8807 pts/1    S+     0:00 grep hd-idle
```

Will post results.