openwrt/packages #11358: cryptsetup/lvm2: Unable to open encrypted partition

aut0, 2020-02-16:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @dangowrt 
Environment: mips, TP-Link Archer C7 v4, OpenWrt 19.07.1 r10911-c155900f66 / LuCI openwrt-19.07 branch git-20.029.45734-adbbd5c

Description: I am trying to open an encrypted partition like so:
```
root@OpenWrt:~# cryptsetup -v --debug open /dev/sda1 testtest
# cryptsetup 2.1.0 processing "cryptsetup -v --debug open /dev/sda1 testtest"
# Running command open.
# Locking memory.
# Installing SIGINT/SIGTERM handler.
# Unblocking interruption on signal.
# Allocating context for crypt device /dev/sda1.
# Trying to open and read device /dev/sda1 with direct-io.
# Initialising device-mapper backend library.
# Trying to load any crypt type from device /dev/sda1.
# Crypto backend (Linux 4.14.167 kernel cryptoAPI) initialized in cryptsetup library version 2.1.0.
# Detected kernel Linux 4.14.167 mips.
# PBKDF pbkdf2, hash sha256, time_ms 2000 (iterations 0), max_memory_kb 0, parallel_threads 0.
# Reading LUKS header of size 1024 from device /dev/sda1
# Key length 32, device size 7814035086 sectors, header size 2050 sectors.
# Activating volume testtest using token -1.
# Interactive passphrase entry requested.
Enter passphrase for /dev/sda1: 
# Activating volume testtest [keyslot -1] using passphrase.
# dm version   [ opencount flush ]   [16384] (*1)
# dm versions   [ opencount flush ]   [16384] (*1)
# Detected dm-ioctl version 4.37.0.
# Detected dm-crypt version 1.18.1.
# Device-mapper backend running with UDEV support disabled.
# dm status testtest  [ opencount noflush ]   [16384] (*1)
# Trying to open key slot 0 [ACTIVE_LAST].
# Releasing crypt device /dev/sda1 context.
# Releasing device-mapper backend.
# Unlocking memory.
Command failed with code -1 (wrong or missing parameters).
```

When i strace the call I see the following:
```
writev(1, [{iov_base="# Activating volume testtest [ke"..., iov_len=59}, {iov_base="\n", iov_len=1}], 2# Activating volume testtest [keyslot -1] using passphrase.
) = 60
uname({sysname="Linux", nodename="OpenWrt", ...}) = 0
stat64("/dev/mapper/control", {st_mode=S_IFCHR|0600, st_rdev=makedev(0xa, 0xec), ...}) = 0
open("/dev/mapper/control", O_RDWR|O_LARGEFILE) = 5
open("/proc/devices", O_RDONLY|O_LARGEFILE) = 6
read(6, "Character devices:\n  1 mem\n  4 t"..., 1024) = 367
_llseek(6, -11, [356], SEEK_CUR)        = 0
close(6)                                = 0
brk(0x4f9000)                           = 0x4f9000
writev(1, [{iov_base="# dm version   [ opencount flush"..., iov_len=49}, {iov_base="\n", iov_len=1}], 2# dm version   [ opencount flush ]   [16384] (*1)
) = 50
ioctl(5, DM_VERSION, {version=4.0.0, data_size=16384, flags=DM_EXISTS_FLAG} => {version=4.37.0, data_size=16384, flags=DM_EXISTS_FLAG}) = 0
writev(1, [{iov_base="# dm versions   [ opencount flus"..., iov_len=50}, {iov_base="\n", iov_len=1}], 2# dm versions   [ opencount flush ]   [16384] (*1)
) = 51
ioctl(5, DM_LIST_VERSIONS, {version=4.1.0, data_size=16384, data_start=312, flags=DM_EXISTS_FLAG} => {version=4.37.0, data_size=456, data_start=312, flags=DM_EXISTS_FLAG, ...}) = 0
writev(1, [{iov_base="# Detected dm-ioctl version 4.37"..., iov_len=35}, {iov_base="\n", iov_len=1}], 2# Detected dm-ioctl version 4.37.0.
) = 36
writev(1, [{iov_base="# Detected dm-crypt version 1.18"..., iov_len=35}, {iov_base="\n", iov_len=1}], 2# Detected dm-crypt version 1.18.1.
) = 36
writev(1, [{iov_base="# Device-mapper backend running "..., iov_len=59}, {iov_base="\n", iov_len=1}], 2# Device-mapper backend running with UDEV support disabled.
) = 60
writev(1, [{iov_base="# dm status testtest  [ opencoun"..., iov_len=58}, {iov_base="\n", iov_len=1}], 2# dm status testtest  [ opencount noflush ]   [16384] (*1)
) = 59
ioctl(5, DM_TABLE_STATUS, {version=4.0.0, data_size=16384, data_start=312, name="testtest", flags=DM_EXISTS_FLAG|DM_NOFLUSH_FLAG} => {version=4.37.0, data_size=16384, data_start=312, name="testtest", flags=DM_EXISTS_FLAG|DM_NOFLUSH_FLAG}) = -1 ENXIO (No such device or address)
writev(1, [{iov_base="# Trying to open key slot 0 [ACT"..., iov_len=42}, {iov_base="\n", iov_len=1}], 2# Trying to open key slot 0 [ACTIVE_LAST].
) = 43
```

The DM_TABLE_STATUS ioctl fails with ENXIO. Something goes wrong when creating the /dev/mapper/ file I think. I am not sure. I also see the following error when executing lvm commands:
```
root@OpenWrt:~# /sbin/lvm vgscan
  Reading all physical volumes.  This may take a while...
  Failed to set up async io, using sync io.
```

Best regards,
Lukas

neheb, 2020-02-16:
I will note that cryptsetup depends on several kmod-crypto modules. I made this change for size reasons.

aut0, 2020-02-16:
> I will note that cryptsetup depends on several kmod-crypto modules. I made this change for size reasons.

I have the following installed:
```
root@OpenWrt:~# opkg list-installed | grep kmod-crypto
kmod-crypto-aead - 4.14.167-1
kmod-crypto-crc32c - 4.14.167-1
kmod-crypto-ecb - 4.14.167-1
kmod-crypto-gf128 - 4.14.167-1
kmod-crypto-hash - 4.14.167-1
kmod-crypto-hmac - 4.14.167-1
kmod-crypto-iv - 4.14.167-1
kmod-crypto-manager - 4.14.167-1
kmod-crypto-misc - 4.14.167-1
kmod-crypto-null - 4.14.167-1
kmod-crypto-pcompress - 4.14.167-1
kmod-crypto-rng - 4.14.167-1
kmod-crypto-sha256 - 4.14.167-1
kmod-crypto-user - 4.14.167-1
kmod-crypto-wq - 4.14.167-1
kmod-crypto-xts - 4.14.167-1
```

aut0, 2020-02-16:
I was missing kmod-crypto-sha1.
Thank you for your quick reply.

closed 2020-02-16 by aut0