openwrt/packages #11359: nginx listen port

lucize, 2020-02-16:
Maintainer @peter-stadler 
With the new updates can't find a way to start nginx on port 81 so haproxy can serve it, is port 80 hardcoded somewhere ?

peter-stadler, 2020-02-16:
For your purpose, I would suggest to create a `server.conf` file in `/etc/nginx/conf.d/` containing a `listen 81` directive (that will act as default_server since it is the only one on this port):
```
cat >/etc/nginx/conf.d/server.conf <<EOF
server {
    listen 81;
    listen [::]:81; # for IPv6
    server_name _;
    # ...
}
EOF
```

To answer the question:

For nginx-ssl there are `listen 80` and `listen [::]:80` directives in `/etc/nginx/conf.d/_redirect2ssl.conf`. 

Else the default_server in `/etc/nginx/conf.d/_lan.conf` includes listen directives from `/var/lib/nginx/lan.listen.default`. That file is dynamically created and contains listen directives on port 80 for all local IPs, typically `listen 192.168.1.1:80 default_server;`.

I wrote more details in the [wiki](https://openwrt.org/docs/guide-user/services/webserver/nginx#new_server_parts).

lucize, 2020-02-16:
> Else the default_server in /etc/nginx/conf.d/_lan.conf

this is the problem, it always generates 80 and I added listen to all server entries

peter-stadler, 2020-02-16:
If you do not want the default server `_lan.conf` comment or remove at least
```
	include '/var/lib/nginx/lan.listen.default';
	server_name _lan;
```
(If you remove the file it will be re-installed when updating nginx.)

peter-stadler, 2020-02-16:
(For nginx without SSL it is enough to comment/remove the include)

peter-stadler, 2020-02-16:
How did you start Nginx on port 81 before. Maybe there is even a better solution.

lucize, 2020-02-16:
ok, so there is not way to generate the file via a variable ? if not maybe a warning in the file
before the update, I just replaced 80 with 81 in the main file, now it tries 80 and 81 despite commenting 	`include '/var/lib/nginx/lan.listen.default';` but at least it's starting with the comment

peter-stadler, 2020-02-16:
I am working on making the default_server configurable via uci, then it will use a variable ...
It should work if you comment this line AND add `listen 81; listen [::]:81;` instead (or in another server part as suggested above), isn't it?

peter-stadler, 2020-02-16:
I see thank you for your suggestion to add the warning :-)

peter-stadler, 2020-02-16:
(If it is working you can restrict the listen directive to the needed IP using `listen 127.0.0.1:81` for example.)

peter-stadler, 2020-02-16:
If I understood you right you do not have the file `/etc/nginx/conf.d/_redirect2ssl.conf`, do you? By default there should be no other configuration listening on port 80.

You can show the whole configuration (from all used files) by `nginx -T`. Are there other `listen` directives than on port 81? (Nginx listens per default on port 80.)

lucize, 2020-02-16:
don't use ssl
attached the -T output (edited de fqdn)
[nginx_conf.txt](https://github.com/openwrt/packages/files/4210864/nginx_conf.txt)

peter-stadler, 2020-02-16:
I copied your config restarted nginx and `netstat -ntulpe` is showing me that it is listening only on `0.0.0.0:81`. So, it should work ...

Beside that, there are some unused parts: Now the new server in `_lan.conf` will not be used since you have another `default_server` in the main configuration.

If you want to use the new defaults you could structure it in the following way:

1) Use the [main configuration](https://raw.githubusercontent.com/openwrt/packages/master/net/nginx/files/nginx.conf) by `mv /etc/nginx/nginx.conf-opkg /etc/nginx/nginx.conf`. This file is not changed by `opkg update nginx` if it was modified. 
By using the upstream version you will get updates. (If you do not want that maybe remove also the `include /etc/nginx/conf.d/*.conf;` line instead.)

2) Move your custom server into the `/etc/nginx/conf.d/` directory.
In your case the first one can be omitted as it is served by the modified `_lan.conf` (you could add `default_server` or not, it could chang only something if you would have another server with an invalid server_name).
 For the second one (mail) you can use:
```
cat > /etc/nginx/conf.d/mail.conf <<EOF
server {
    listen 81 ;
    server_name mail.<domain>;

    location /webmail {
        return 301 https://$server_name$request_uri;
    }
    location /.well-known {
        include         conf.d/proxy_set_header.inc;
        proxy_pass      http://192.168.68.254/.well-known;
    }

    location ~* .(jpg|jpeg|png|gif|ico|css|js)$ {
        expires 365d;
    }

    include luci_uwsgi.conf;
}
EOF
```
I suppose the https for the `location /webmail` is handled somehow else ...
And if you do not want LuCI to be reachable by mail.<domain>, remove the include and maybe also
```
    location ~* .(jpg|jpeg|png|gif|ico|css|js)$ {
        expires 365d;
    }
```

3) If you want to delete the `/etc/nginx/luci_uwsgi.conf` file, replace the include by
`include /etc/nginx/conf.d/luci.locations;` (if still there)

closed 2020-02-17 by lucize