openwrt/packages #11369: samba4-server: macOS cannot write to network share

jediwade, 2020-02-18:
Maintainer: @Andy2244
Environment: ARMv7 Processor rev 1 (v7l), Linksys WRT3200ACM, OpenWrt 19.07.1 r10911-c155900f66 / LuCI openwrt-19.07 branch git-20.046.57886-0da47ff

Description: Cannot copy files to or modify files on Samba4 network share from Macbook Air (2013) running macOS Sierra (10.12.6). Can successfully read files (watch movie, open text document). On a Windows 10 computer, can successfully write and copy to network share.

Network share is 3TB WD NAS formatted as ExFAT connected through USB3.

Receive "(Error code -36)" when attempting to copy or modify files. This issue is reported as fixed in Samba 4.2.8 (https://wiki.samba.org/index.php/Samba_4.2_Features_added/changed). Search for "macos" and this issue is the only result.

Andy2244, 2020-02-18:
I assume you have the "MacOS" feature enabled in luci? Did you also try enabling the "Force root" option?

jediwade, 2020-02-18:
Yes, I have checked the "Enable macOS compatible shares" in luci. I also
tried the "Force root" option and restarted the router after making that
change. I will try it again tonight after I get home from work. I also
manually changed the "Create mask" to "0744" and "Directory mask" to "0777"
options to match the default according to Samba documentation I found. I do
not have the "Apple Time-machine share" option selected because I do not
care about Time-machine backups.

Andy2244, 2020-02-18:
>  I also manually changed the "Create mask" to "0744" and "Directory mask" to "0777"

Also make sure that your share mount folder has 0777 rights, so as example `/mnt/sda1`

jediwade, 2020-02-18:
> > I also manually changed the "Create mask" to "0744" and "Directory mask" to "0777"
> 
> Also make sure that your share mount folder has 0777 rights, either via shell or sftp tools, so as example `/mnt/sda1`. Via sftp you can also test if the basic read/writes work as expected, so we can exclude exfat kmod problems.

Forgive my ignorance on the matter, but how would I do that? I am a casual user of OpenWRT who just wanted to breathe new life into a router that Linksys doesn't seem to support anymore. Can this be done through LuCI or does it have to be done through shell? My mount matches the example you gave (/mnt/sda1).

I also tested the "Force root" option again and restarted the router after saving/applying the change, but the result is the same.

Andy2244, 2020-02-18:
In the mac "terminal" type this and use your router password.
```
ssh rooty@192.168.1.1
chmod 0777 /mnt/sda1
exit
```

jediwade, 2020-02-18:
For a few minutes after restarting the router, I couldn't even reconnect. I was eventually able to but the result is the same.

Not sure that if it matters or not, but I have just been using the "Allow guests" option and have not made any user/pass requirements. For my network share, I have the following options set:

GENERAL SETTINGS:
Interface: lan
Workgroup: WORKGROUP
Description: Samba on OpenWrt
Force synchronous I/O: NOT Checked
Enable macOS compatible shares: checked
Allow legacy (insecure) protocols/auth: NOT checked
  
SHARED DIRECTORIES:
Name: Jade Shadow
Path: /mnt/sda1
Browse-able: checked
Read-only: NOT checked
Force root: checked
Allowed users: [blank]
Allow guests: checked
Guest only: NOT checked
Inherit owner: checked
Create mask: 0744
Directory mask: 0777
Vfs objects: [blank]
Apple Time-machine share: NOT checked
Time-machine size in GB: [blank]

Thank you for all your help in trying to fix this.

Andy2244, 2020-02-19:
Can you try create a new test share on the router memory filesystem itself at `/tmp` and than check if you can write files and create folders there? This way we can quickly check if its a exfat vs native linux filesystem problem.

jediwade, 2020-02-20:
I created the new share as you suggested pointing to "/tmp" (having the same options above set) and it is the same result. The exact error message is:

The Finder can't complete the operation because some data in "[filename]" can't be read or written. (Error code -36)

One thing I forgot to mention is that it does create a file on the network share, but it is zero bytes in size.

Andy2244, 2020-02-20:
mhh ok, guess i need to reinstall VMware and see if my old macOS VM still boots and i can recreate the issue.

The only thing left i can think of, did you try disable the "macOS" option? Maybe some of the VFS modules i add for mac is not compatible anymore.

Andy2244, 2020-02-20:
ah btw do you see any `daemon.err smbd` error messages in your system log, other than the startup message?

jediwade, 2020-02-20:
Well that just seems silly. Turning off the "Enable macOS" option fixed it. Before I did that, I tried to copy a file over so you could have an error log.

Thu Feb 20 06:27:33 2020 daemon.err uhttpd[2409]: luci: accepted login on / for root from 192.168.1.232
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]: [2020/02/20 06:28:07.068171,  0] ../../source3/modules/vfs_fruit.c:1152(ad_convert_xattr)
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]:   ad_convert_xattr: SMB_VFS_CREATE_FILE failed
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]: [2020/02/20 06:28:07.068729,  0] ../../source3/modules/vfs_fruit.c:6129(fruit_readdir_attr)
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]:   fruit_readdir_attr: ad_convert() failed
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]: [2020/02/20 06:28:07.080166,  0] ../../source3/modules/vfs_fruit.c:1152(ad_convert_xattr)
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]:   ad_convert_xattr: SMB_VFS_CREATE_FILE failed
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]: [2020/02/20 06:28:07.080329,  0] ../../source3/modules/vfs_fruit.c:6129(fruit_readdir_attr)
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]:   fruit_readdir_attr: ad_convert() failed
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]: [2020/02/20 06:28:07.091123,  0] ../../source3/modules/vfs_fruit.c:1152(ad_convert_xattr)
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]:   ad_convert_xattr: SMB_VFS_CREATE_FILE failed
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]: [2020/02/20 06:28:07.091278,  0] ../../source3/modules/vfs_fruit.c:6129(fruit_readdir_attr)
Thu Feb 20 06:28:07 2020 daemon.err smbd[6800]:   fruit_readdir_attr: ad_convert() failed
Thu Feb 20 06:28:11 2020 daemon.err smbd[6800]: [2020/02/20 06:28:11.540589,  0] ../../source3/modules/vfs_fruit.c:4214(fruit_pread_meta_stream)
Thu Feb 20 06:28:11 2020 daemon.err smbd[6800]:   fruit_pread_meta_stream: Removing [file-alt.ai:AFP_AfpInfo] after short read [0]
Thu Feb 20 06:28:11 2020 daemon.err smbd[6800]: [2020/02/20 06:28:11.544908,  0] ../../source3/modules/vfs_fruit.c:4557(fruit_pwrite_meta_stream)
Thu Feb 20 06:28:11 2020 daemon.err smbd[6800]:   fruit_pwrite_meta_stream: On-demand create [file-alt.ai:AFP_AfpInfo] in write failed: No such file or directory

Andy2244, 2020-02-20:
> Turning off the "Enable macOS" option fixed it.

Yeah in principle macOS can use normal Samba shares, yet it want's to store some special access, file data and the MacOS option adds this feature via [VFS modules](https://www.samba.org/samba/docs/current/man-html/vfs_fruit.8.html) _(fruit, xattr)_, so the extra data gets stored.
This worked fine, but seems Samba4.11 broke "stuff" and newer macOS versions also handle things slightly different now. So needs to-be fixed by the samba team.

If i have some extra time, i will try figure out why fruit + xattr is not working anymore and check if the alternative modes work.

poranje, 2020-02-21:
FYI: on 19.07.01 ipq806x (netgear,r7800), with samba4 with option macos '1' and a share on ext4, reading and writing from macOS 10.15 to that share works well.

gfrancesco, 2020-03-31:
If of any help:
- Samba 4.11.6
- OpenWrt 19.07.1 r10911-c155900f66
- Feroceon 88FR131 rev 1 (v5l) (ZyXEL NSA325 NAS)
- MacOS 10.15.4
- Several XFS backed shares
- In Samba NO guest enabled, NO macos option enabled

Never had any problem reading/writing any of the share, once file permissions on OpenWrt and Samba are set correctly.

alst-ovh, 2020-04-15:
Have here the same Problem.
Works with mounted ext shares bur not on fat or exfat.
Copy files with windows works, but not with MacOs Finder?

Apr 15 18:17:12 odw-cmd smbd[1117]: [2020/04/15 18:17:12.463851,  0] ../source3/modules/vfs_fruit.c:4252(fruit_pread_meta_stream)
Apr 15 18:17:12 odw-cmd smbd[1117]:   fruit_pread_meta_stream: Removing [sde1.vfat/DA-71117_manual_de_German_20190401.pdf:AFP_AfpInfo] after short read [0]
Apr 15 18:17:12 odw-cmd smbd[1117]: [2020/04/15 18:17:12.464930,  0] ../source3/modules/vfs_fruit.c:4595(fruit_pwrite_meta_stream)
Apr 15 18:17:12 odw-cmd smbd[1117]:   fruit_pwrite_meta_stream: On-demand create [sde1.vfat/DA-71117_manual_de_German_20190401.pdf:AFP_AfpInfo] in write failed: Datei oder Verzeichnis nicht gefunden
Apr 15 18:17:24 odw-cmd smbd[1117]: [2020/04/15 18:17:24.432244,  0] ../source3/modules/vfs_fruit.c:4252(fruit_pread_meta_stream)
Apr 15 18:17:24 odw-cmd smbd[1117]:   fruit_pread_meta_stream: Removing [DWBackup_1.ext4/DA-71117_manual_de_German_20190401.pdf:AFP_AfpInfo] after short read [0]
Apr 15 18:17:24 odw-cmd smbd[1117]: [2020/04/15 18:17:24.432337,  0] ../source3/modules/vfs_fruit.c:4256(fruit_pread_meta_stream)
Apr 15 18:17:24 odw-cmd smbd[1117]:   fruit_pread_meta_stream: Removing [DWBackup_1.ext4/DA-71117_manual_de_German_20190401.pdf:AFP_AfpInfo] failed
Apr 15 18:17:44 odw-cmd smbd[1117]: [2020/04/15 18:17:44.564782,  0] ../source3/modules/vfs_fruit.c:4252(fruit_pread_meta_stream)
Apr 15 18:17:44 odw-cmd smbd[1117]:   fruit_pread_meta_stream: Removing [DWBackup_1.ext4/DA-71117_manual_de_German_20190401.pdf:AFP_AfpInfo] after short read [0]
Apr 15 18:17:44 odw-cmd smbd[1117]: [2020/04/15 18:17:44.564875,  0] ../source3/modules/vfs_fruit.c:4256(fruit_pread_meta_stream)
Apr 15 18:17:44 odw-cmd smbd[1117]:   fruit_pread_meta_stream: Removing [DWBackup_1.ext4/DA-71117_manual_de_German_20190401.pdf:AFP_AfpInfo] failed

gfrancesco, 2020-04-15:
Is macOS option enabled?

alst-ovh, 2020-04-19:
ea support = Yes
Dit you mean this ?

gfrancesco, 2020-04-19:
That's only part of it, see here: https://wiki.samba.org/index.php/Configure_Samba_to_Work_Better_with_Mac_OS_X

I have none of that enabled and it works fine, you need to have `SMB2` as minimum protocol but that should be a default, same for `ea support` depending on your Samba version (see link above).
Check also on your Luci conf page for Samba if you have the `Enable macOS compatible shares` ticked.

ffainelli, 2020-10-24:
Also encountering the same issue here and so far have not been able to find out what is the reason for that. My set-up is a bit more complicated since the Time machine location is SMB mounted from another machine and then re-exported via a samba4 installation on another OpenWrt device.

julienplee, 2021-03-02:
> Have here the same Problem.
> Works with mounted ext shares bur not on fat or exfat.
> Copy files with windows works, but not with MacOs Finder?

Hello,
Your problem should be a bit different than the current subject.
This is because FAT nor exFAT doesn't support extended attributes (data that allow to put comments or tags on files, remember if a directory is open or closed…).
Those attributes can be stored on files starting with '._' followed by the name of the file. This is how OSX handles an ex/FAT drive directly plugged. But you have to tell Samba not to try to write in XATTR area of the file system, at least for those disks (EXT4 has an XATTR support, but not as large as OSX and you may encounter problems one day). 

If you encounter one of the following commented option in your Samba global conf, please change it for the uncommented version. Samba allows you to apply those options either globally or per share. Because you might have one permanently plugged EXT drive and often plug FAT/exFAT drives, I suggest you add the existing options on the EXT share (so that it continues to write XATTRs on the filesystem XATTR, and apply globally the XATTR in '._' files, so that you don't have to worry about the file system support.

```
#fruit:resource = stream  or  xattr
fruit:resource = file
#fruit:metadata = stream
fruit:metadata = netatalk
fruit:veto_appledouble = yes    # (prevents Windows to list the OSX metadata files)
```
I guess this is all you need.

Andy2244, 2021-03-02:
> This may be because neither FAT nor exFAT support extended attributes (data that allow to put comments or tags on files, remember if a directory is open or closed…).
> When not supported by the filesystem, those attributes can be stored on files starting with '._' followed by the name of the file. This is how OSX handles an ex/FAT drive directly plugged. But you have to tell Samba not to try to write in XATTR area of the file system, at least for those disks (EXT4 has an XATTR support, but not as large as OSX and you may encounter problems one day).

The default values we use in openwrt is to store any extra metadata/streams in a xattr database file and not use any filesystem specific xattr values. Thats why we have set `vfs_objects= catia fruit streams_xattr` with `fruit:metadata = stream`. This should pass any metadata to the `streams_xattr` module, which should handle all the extra metadata stuff even on a filesystem like FAT.

jediwade, 2021-03-03:
My apologies for leaving this open after it was resolved. For others needing this for future reference, listed below are my settings to have this work with macOS (leaving out custom values such as description, path, name, workgroup, etc).

GENERAL SETTINGS
Interface: lan
Force sync: unchecked
Enable macOS: unchecked
Allow legacy: unchecked
Disable netbios: checked

EDIT TEMPLATE
## gives network share the Apple Timecapsule appearance in Finder
fruit:model = TimeCapsule8,119

SHARE DIRECTORIES
Browse-able: checked
Read-only: unchecked
Force root: checked
Allow guests: unchecked (users configured)
Guests only: unchecked
Inherit owner: unchecked
Create mask: 0744
Directory mask: 0755
Vfs objects: (blank)
Apple Time-machine share: unchecked
Time-machine size: (blank)

closed 2021-03-03 by jediwade