openwrt/packages #11377: libffmpeg-full is missing dependencies for the following libraries: libpostproc.so.55

feizaibufei, 2020-02-18:
mkdir -p "/home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/lib/pkgconfig"
install -m 644 libavutil/libavutil.pc "/home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/lib/pkgconfig"
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2'
touch /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.built
rm -rf /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg.installed /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg
mkdir -p /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg
install -d -m0755 /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg/usr/bin
cp -fpR /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/bin/ffmpeg /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg/usr/bin/
touch /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg.installed
mkdir -p /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/root-armvirt/stamp
SHELL= flock /home/openwrt/lede/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg/. /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/root-armvirt/'
touch /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/root-armvirt/stamp/.ffmpeg_installed
mkdir -p /home/openwrt/lede/bin/targets/armvirt/64/packages /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full/CONTROL /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/pkginfo
install -d -m0755 /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full/usr/lib
cp -fpR /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/lib/lib{avcodec,avdevice,avfilter,avformat,avresample,avutil,swresample,swscale}.so.* /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full/usr/lib/
find /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full -name 'CVS' -o -name '.svn' -o -name '.#' -o -name '~'| xargs -r rm -rf
Package libffmpeg-full is missing dependencies for the following libraries:
libpostproc.so.55
make[3]: *** [Makefile:808: /home/openwrt/lede/bin/packages/aarch64_generic/packages/libffmpeg-full_4.2.2-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/openwrt/lede/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#545.18#77.51#620.68
make[2]: *** [package/Makefile:109: package/feeds/packages/ffmpeg/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [package/Makefile:103: /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:218：world] Error 2

neheb, 2020-02-19:
ping @diizzyy

diizzyy, 2020-02-19:
@feizaibufei 
Did you change anything in the Makefile or enable a specific option?

feizaibufei, 2020-02-19:
[image]
FFmpeg compiled successfully

neheb, 2020-02-22:
I take it this is solved?

neheb, 2020-03-09:
Closing due to timeout.

closed 2020-03-09 by neheb

phantnang, 2024-06-23:
please help me
can you commpile libffmpeg-full for my new router xiaomi r3g migsel-24kc on openwrt 19.07.10

neheb, 2024-06-23:
19.07 is EOL

phantnang, 2024-06-24:
any newer is ok.
my older router run v15 with ffmpeg 2.6, it have libpostproc.