openwrt/packages #11384: collect2: error: ld returned 1 exit status

snakwu, 2020-02-19:
time: package/libs/gmp/compile#0.19#0.10#0.30
make[3]: Entering directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/utils/coreutils'
rm -f /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/.built
touch /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/.built_check
make -C /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31 DESTDIR="/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/ipkg-install" SHELL="/bin/bash" all install
make[4]: Entering directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31'
make  all-recursive
make[5]: Entering directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31'
Making all in po
make[6]: Entering directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/po'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/po'
make[6]: Entering directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31'
x86_64-openwrt-linux-musl-gcc   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31=coreutils-8.31 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wl,--as-needed -L/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/lib -L/home/snakwu/openwrt-master/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/snakwu/openwrt-master/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro  -o src/od src/od.o src/libver.a lib/libcoreutils.a  lib/libcoreutils.a 
/home/snakwu/openwrt-master/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: lib/libcoreutils.a(strtold.o):(.data+0x0): multiple definition of `minus_zero'; lib/libcoreutils.a(strtod.o):(.data+0x0): first defined here
collect2: error: ld returned 1 exit status
Makefile:8532: recipe for target 'src/od' failed
make[6]: *** [src/od] Error 1
make[6]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31'
Makefile:12161: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31'
Makefile:6347: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31'
Makefile:144: recipe for target '/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/.built' failed
make[3]: *** [/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.31/.built] Error 2
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/utils/coreutils'
time: package/feeds/packages/coreutils/compile#0.78#0.15#1.15
package/Makefile:111: recipe for target 'package/feeds/packages/coreutils/compile' failed
make[2]: *** [package/feeds/packages/coreutils/compile] Error 2
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
package/Makefile:107: recipe for target '/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

garypang13, 2020-02-19:
Same error with me

garypang13, 2020-02-19:
@snakwu 应该是最新的这个https://github.com/openwrt/packages/commit/89d0510d2ed35e92fc64e1d2b304cfd83167737dcommit 造成的.

snakwu, 2020-02-19:
> @snakwu 应该是最新的这个[7b59b78](https://github.com/openwrt/packages/commit/7b59b783f027f9b0fcc57d1d697361dc87c5dbd7)
> commit 造成的.

可有解决办法？

garypang13, 2020-02-19:
> > @snakwu 应该是最新的这个[7b59b78](https://github.com/openwrt/packages/commit/7b59b783f027f9b0fcc57d1d697361dc87c5dbd7)
> > commit 造成的.
> 
> 可有解决办法？

coreutils回退到上个commit,我还在测试编译中,

snakwu, 2020-02-19:
ok，等你结果，谢谢！

feckert, 2020-02-19:
I don't think it's very fair! Please talk in English.
I have the same problem on my build!

garypang13, 2020-02-19:
> ok，等你结果，谢谢！

@snakwu  成功了 在执行 make -j 前 在根目录执行
`svn co --force https://github.com/openwrt/packages/branches/openwrt-19.07/utils/coreutils feeds/packages/utils/coreutils && svn revert -R feeds/packages/utils/coreutils`
即可

snakwu, 2020-02-19:
> > ok，等你结果，谢谢！
> 
> @snakwu 成功了 在执行 make -j 前 在根目录执行
> `svn co --force https://github.com/openwrt/packages/branches/openwrt-19.07/utils/coreutils feeds/packages/utils/coreutils && svn revert -R feeds/packages/utils/coreutils`
> 即可

明白了,谢谢你!果然是早上的那个更新引起的!

garypang13, 2020-02-19:
> I don't think it's very fair! Please talk in English.
> I have the same problem on my build!

@feckert Sorry, run `svn co --force https://github.com/openwrt/packages/branches/openwrt-19.07/utils/coreutils feeds/packages/utils/coreutils && svn revert -R feeds/packages/utils/coreutils` before you run make -j, It can fix this problem.

neheb, 2020-02-21:
@garypang13 svn. interesting...

AFAIK, I got rid of all svn usage in here.

snakwu, 2020-02-21:
> > ok，等你结果，谢谢！
> 
> @snakwu 成功了 在执行 make -j 前 在根目录执行
> `svn co --force https://github.com/openwrt/packages/branches/openwrt-19.07/utils/coreutils feeds/packages/utils/coreutils && svn revert -R feeds/packages/utils/coreutils`
> 即可

你好，之前用了svn命令控制了版本，现在fix了，请问怎么解除svn的版本控制？

yousong, 2020-02-21:
> > > ok，等你结果，谢谢！
> > 
> > 
> > @snakwu 成功了 在执行 make -j 前 在根目录执行
> > `svn co --force https://github.com/openwrt/packages/branches/openwrt-19.07/utils/coreutils feeds/packages/utils/coreutils && svn revert -R feeds/packages/utils/coreutils`
> > 即可
> 
> 你好，之前用了svn命令控制了版本，现在fix了，请问怎么解除svn的版本控制？现在版本让控制在8.30，引起fix不适用编译错误

English please, so that others can follow

snakwu, 2020-02-21:
> English please, so that others can follow

Hello, I used the svn command to restore the version, but now it has been repaired. How can I remove svn, from using the latest files? The current version is controlled at 8.30, which causes a compilation error that fix is not applicable.

yousong, 2020-02-21:
> Hello, I used the svn command to restore the version, but now it has been repaired. How can I remove svn, from using the latest files? The current version is controlled at 8.30, which causes a [...]

Not sure why svn was used here.  If you cannot find a better way to restore to using current upstream version, just remove `feeds/packages`, then `./scripts/feeds update`, etc.

snakwu, 2020-02-21:
> > > > > > ok，等你结果，谢谢！
> > > > > 
> > > > > 
> > > > > @snakwu 成功了 在执行 make -j 前 在根目录执行
> > > > > `svn co --force https://github.com/openwrt/packages/branches/openwrt-19.07/utils/coreutils feeds/packages/utils/coreutils && svn revert -R feeds/packages/utils/coreutils`
> > > > > 即可
> > > > 
> > > > 
> > > > 你好，之前用了svn命令控制了版本，现在fix了，请问怎么解除svn的版本控制？现在版本让控制在8.30，引起fix不适用编译错误
> > > 
> > > 
> > > English please, so that others can follow
> > 
> > 
> > Hello, I used the svn command to restore the version, but now it has been repaired. How can I remove svn, from using the latest files? The current version is controlled at 8.30, which causes a compilation error that fix is not applicable.
> 
> Not sure why svn was used here. If you cannot find a better way to restore to using current upstream version, just remove `feeds/packages` (save your working modifications of course!), then `./scripts/feeds update`, etc.

ok，thx！

yousong, 2020-02-21:
Fixed by pull request #11391

closed 2020-02-21 by yousong