openwrt/packages #11387: cryptsetup - doesn't create /dev/mapper/ device - can't mount enc

mwoz123, 2020-02-19:
Maintainer: @dangowrt 
Environment:
```
root@OpenWrt:~# uname -a
Linux OpenWrt 4.14.167 #0 Wed Jan 29 16:05:35 2020 mips GNU/Linux
root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="19.07.1"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.1"
VERSION_ID="19.07.1"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r10911-c155900f66"
OPENWRT_BOARD="ath79/generic"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.1 r10911-c155900f66"
```

Description:

```
root@OpenWrt:~# opkg install kmod-crypto-ecb kmod-crypto-xts kmod-crypto-iv kmod-crypto-misc kmod-crypto-user cryptsetup
Installing kmod-crypto-ecb (4.14.167-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/targets/ath79/generic/kmods/4.14.167-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-crypto-ecb_4.14.167-1_mips_24kc.ipk
Installing kmod-crypto-xts (4.14.167-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/targets/ath79/generic/kmods/4.14.167-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-crypto-xts_4.14.167-1_mips_24kc.ipk
Installing kmod-crypto-iv (4.14.167-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/targets/ath79/generic/kmods/4.14.167-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-crypto-iv_4.14.167-1_mips_24kc.ipk
Installing kmod-crypto-misc (4.14.167-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/targets/ath79/generic/kmods/4.14.167-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-crypto-misc_4.14.167-1_mips_24kc.ipk
Installing kmod-crypto-user (4.14.167-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/targets/ath79/generic/kmods/4.14.167-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-crypto-user_4.14.167-1_mips_24kc.ipk
Installing cryptsetup (2.1.0-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mips_24kc/packages/cryptsetup_2.1.0-2_mips_24kc.ipk
Configuring kmod-crypto-user.
Configuring kmod-crypto-xts.
Configuring kmod-crypto-misc.
Configuring kmod-crypto-iv.
Configuring cryptsetup.
Configuring kmod-crypto-ecb.
root@OpenWrt:~# opkg install kmod-fs-ext4 e2fsprogs
Installing kmod-fs-ext4 (4.14.167-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/targets/ath79/generic/kmods/4.14.167-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d/kmod-fs-ext4_4.14.167-1_mips_24kc.ipk
Installing e2fsprogs (1.44.5-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mips_24kc/base/e2fsprogs_1.44.5-2_mips_24kc.ipk
Configuring kmod-fs-ext4.
Configuring e2fsprogs.
root@OpenWrt:~# cryptsetup open /dev/sdb4 crypt
Enter passphrase for /dev/sdb4: 
root@OpenWrt:~# ls /dev/mapper/
control
root@OpenWrt:~# 

```
no `/dev/mapper/`device is created so I can't mount that, neither error.
```
                                                      Disk: /dev/sdb
                                Size: 931.49 GiB, 1000170586112 bytes, 1953458176 sectors
                               Label: gpt, identifier: 709B79B8-20F8-4DB9-B393-BD4C90E93895

    Device                          Start                End            Sectors           Size Type
    /dev/sdb1                        2048            1050623            1048576           512M Microsoft basic data
[... 88 lines cut ...]
# dm resume sdc4_crypt  [ opencount flush securedata ]   [16384] (*1)
# sdc4_crypt: Stacking NODE_ADD (253,2) 0:6 0660 [trust_udev]
# sdc4_crypt: Stacking NODE_READ_AHEAD 256 (flags=1)
# Udev cookie 0xd4dbfd7 (semid 131072) decremented to 1
# Udev cookie 0xd4dbfd7 (semid 131072) waiting for zero
# Udev cookie 0xd4dbfd7 (semid 131072) destroyed
# sdc4_crypt: Skipping NODE_ADD (253,2) 0:6 0660 [trust_udev]
# sdc4_crypt: Processing NODE_READ_AHEAD 256 (flags=1)
# sdc4_crypt (253:2): read ahead is 256
# sdc4_crypt: retaining kernel read ahead of 256 (requested 256)
# Releasing crypt device /dev/sdc4 context.
# Releasing device-mapper backend.
# Unlocking memory.
Polecenie się powiodło.
marcin@pc:~$ ls /dev/mapper/
control  sda2_crypt  sdb3_crypt  sdc4_crypt
```
translation:
`Hasło dla /dev/sdc4:` ~ `Enter passphrase for /dev/sdc4:`
`Klucz numer 0 odblokowany.` ~ `Key number 0 unlocked`
`Polecenie się powiodło.` ~ `Command success`

as far as I see main difference is:
`# Device-mapper backend running with UDEV support enabled.` - pc vs:
`# Device-mapper backend running with UDEV support disabled.` - router

dangowrt, 2020-02-19:
Looks like you may be missing a kernel module needed for opening the passphrase. As cryptsetup now uses the in-kernel crypto instead of linking OpenSSL, you need to also install the kernel modules for the key-hashing algo used on your volume (most likely `kmod-crypto-sha256`).
Please try `cryptsetup benchmark` and see if there are results for all cypher and hash algorithms you are using for that encrypted volume.

mwoz123, 2020-02-20:
I forgot to mention that I also rebooted router but without any changes.
```
root@OpenWrt:~# cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1          N/A
PBKDF2-sha256      35656 iterations per second for 256-bit key
PBKDF2-sha512        N/A
PBKDF2-ripemd160     N/A
PBKDF2-whirlpool   10529 iterations per second for 256-bit key
argon2i       4 iterations,  7828 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
argon2id      4 iterations,  7727 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b               N/A               N/A
    serpent-cbc        128b               N/A               N/A
    twofish-cbc        128b               N/A               N/A
        aes-cbc        256b               N/A               N/A
    serpent-cbc        256b               N/A               N/A
    twofish-cbc        256b               N/A               N/A
        aes-xts        256b         6.3 MiB/s         7.8 MiB/s
    serpent-xts        256b         5.1 MiB/s         6.3 MiB/s
    twofish-xts        256b         6.1 MiB/s         5.0 MiB/s
        aes-xts        512b         5.2 MiB/s         6.2 MiB/s
    serpent-xts        512b         6.2 MiB/s         3.9 MiB/s
    twofish-xts        512b         4.9 MiB/s         5.0 MiB/s
root@OpenWrt:~#
```
can't add kmod-crypto-sha256:
```
root@OpenWrt:~# insmod kmod-crypto-sha256
Failed to find kmod-crypto-sha256. Maybe it is a built in module ?
root@OpenWrt:~# 
```

dangowrt, 2020-02-20:
It'd have been `opkg install kmod-crypto-sha256`, but according to the benchmark output, SHA256 is available and working.
Please provide `cryptsetup luksDump /dev/sdb4`, you may be using some non-standard hash or cypher which would require additional modules.

mwoz123, 2020-02-20:
```
root@OpenWrt:/# cryptsetup luksDump /dev/sdb4
LUKS header information for /dev/sdb4

Version:        1
Cipher name:    aes
Cipher mode:    xts-plain64
Hash spec:      sha1
Payload offset: 4096
MK bits:        256
MK digest:      e0 61 b4 6b e1 97 db 17 d6 be da 1d b8 c1 55 9d bb 84 11 f9 
MK salt:        76 44 d5 b7 4d f7 88 07 51 d2 85 8e 01 4a bd 05 
                30 43 0a 20 b5 f9 2a e1 d6 dd 29 89 68 b0 8c 0e 
MK iterations:  123000
UUID:           a8b119f0-1c38-40ce-b26f-448cddbe350e

Key Slot 0: ENABLED
        Iterations:             496124
        Salt:                   11 b8 0d b2 ab 65 76 31 7d b8 00 fc 41 de 15 8b 
                                ec 6b 42 22 27 f5 e4 88 07 5b 89 77 7f 8a 41 2b 
        Key material offset:    8
        AF stripes:             4000
Key Slot 1: DISABLED
Key Slot 2: DISABLED
Key Slot 3: DISABLED
Key Slot 4: DISABLED
Key Slot 5: DISABLED
Key Slot 6: DISABLED
Key Slot 7: DISABLED
root@OpenWrt:/# 
```
I assume this is most important: 
```
Cipher name:    aes
Cipher mode:    xts-plain64
Hash spec:      sha1
```
I assume I should install something for sha1 + aes + plain xts , is that correct ?

dangowrt, 2020-02-20:
Yes, you are missing the SHA1 kernel module. Please try:
```
opkg install kmod-crypto-sha1
```
and that should allow you to open that volume.
Also be aware that SHA1 is no longer considered safe against powerful enough opponents, see https://hackaday.com/2017/02/23/shattered-sha-1-is-broken/
As an anti-theft-measure it should still be fine for a while, but just saying...

mwoz123, 2020-02-20:
Thanks it worked :)

Before I close it could you tell me (I'm encryption noob and I glance at internet but found nothing) - it's not possible to upgrade the encryption algorithm (from sha1 to sha256) without recreating fs, right?

dangowrt, 2020-02-20:
It's not the encryption cypher but rather the hashing algorithm used to derive the cryptographic key from the password. If you are using enough iterations (496124 looks high enough to me, but don't consider me an expert on cryptography as well) even SHA1 should be safe enough for a while.
Generally, you may want to re-create that volume using the new LUKS2 format which makes attacks much more costly as it not only requires CPU power but also a lot of memory to perform an unlock attempt.

mwoz123, 2020-02-20:
great thanks:)

closed 2020-02-20 by mwoz123