openwrt/packages #11394: Patch failed!  Please fix ./patches/002-strtod-fix-clash-with-strtold.patch!

snakwu, 2020-02-20:
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/package/libs/gmp'      
time: package/libs/gmp/compile#0.17#0.09#0.26                                          
make[3]: Entering directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/utils/c
oreutils'                                                                              
touch /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.30/.
prepared_b8911f361a1cb4e14305166035a28c4f_6664517399ebbbc92a37c5bb081b5c53_check       
. /home/snakwu/openwrt-master/openwrt/include/shell.sh; xzcat /home/snakwu/openwrt-mast
er/openwrt/dl/coreutils-8.30.tar.xz | tar -C /home/snakwu/openwrt-master/openwrt/build_
dir/target-x86_64_musl/coreutils-8.30/.. -xf -                                         
[ ! -d ./src/ ] || cp -fpR ./src/. /home/snakwu/openwrt-master/openwrt/build_dir/target
-x86_64_musl/coreutils-8.30                                                            
                                                                                       
Applying ./patches/001-no_docs_man_tests.patch using plaintext:                        
patching file Makefile.am                                                              
patching file Makefile.in                                                              
                                                                                       
Applying ./patches/002-strtod-fix-clash-with-strtold.patch using plaintext:            
patching file lib/strtod.c                                                             
Hunk #1 FAILED at 294.                                                                 
Hunk #2 succeeded at 341 with fuzz 1 (offset -138 lines).                              
1 out of 2 hunks FAILED -- saving rejects to file lib/strtod.c.rej                     
Patch failed!  Please fix ./patches/002-strtod-fix-clash-with-strtold.patch!           
Makefile:144: recipe for target '/home/snakwu/openwrt-master/openwrt/build_dir/target-x
86_64_musl/coreutils-8.30/.prepared_b8911f361a1cb4e14305166035a28c4f_6664517399ebbbc92a
37c5bb081b5c53' failed                                                                 
make[3]: *** [/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutil
s-8.30/.prepared_b8911f361a1cb4e14305166035a28c4f_6664517399ebbbc92a37c5bb081b5c53] Err
or 1                                                                                   
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/utils/co
reutils'                                                                               
time: package/feeds/packages/coreutils/compile#0.64#0.33#0.92                          
package/Makefile:111: recipe for target 'package/feeds/packages/coreutils/compile' fail
ed                                                                                     
make[2]: *** [package/feeds/packages/coreutils/compile] Error 2                        
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                       
package/Makefile:107: recipe for target '/home/snakwu/openwrt-master/openwrt/staging_di
r/target-x86_64_musl/stamp/.package_compile' failed                                    
make[1]: *** [/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/
.package_compile] Error 2                                                              
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                       
/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:225: recipe for target 'world' 
failed                                                                                 
make: *** [world] Error 2

feckert, 2020-02-20:
I think you have the wrong version -> coreutils-8.30 this is for coreutils-8.31?

> touch /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/coreutils-8.30/.
> prepared_b8911f361a1cb4e14305166035a28c4f_6664517399ebbbc92a37c5bb081b5c53_check
> . [...]

snakwu, 2020-02-20:
I pulled the latest code again, why is that?

snakwu, 2020-02-21:
Ok, resolved

closed 2020-02-21 by snakwu