openwrt/packages #11402: Unbound: trying to compile in ipset support, but ./configure does not find libmnl or libmnl.h

Borromini, 2020-02-21:
Maintainer: @EricLuehrsen 
Environment: master/ath79 (although target is probably not relevant)

Hi Eric! This is not a bug report but a request for help. I'm trying to include [ipset support](https://github.com/NLnetLabs/unbound/pull/28) into the OpenWrt package. From what I can see it should be as easy as adding `+libmnl` as a dependency (which I have added to both the library packages and the main unbound package) and adding `--enable-ipset` to the configure switches in the Makefile. However, a `make package/unbound/{clean,compile}` on master errors out with this error:

`Could not find libmnl, libmnl.h`

There's a `--with-libmnl=path` switch that Unbound configure supports but it seems to be wanting an absolute path, and I'm not sure how to approach that (given the path needs to be in the buildroot).

Can you help me out?

I can file a PR afterwards, I think ipset support would be neat to have both in the default `unbound-daemon` and in `unbound-daemon-heavy`.

Thanks!

EricLuehrsen, 2020-02-22:
Use the makefile staging directory symbol. Unbound flags in use have a few examples:
```
--with-ssl="$(STAGING_DIR)/usr"
```

Borromini, 2020-02-23:
Thanks! That works. Will prep a PR, for unbound-heavy only then.

closed 2020-02-23 by Borromini