openwrt/packages #11410: https-dns-proxy: high CPU/memory usage, crashes

Mrnofish, 2020-02-22:
Maintainer: @stangri 
Environment: mips 24Kc, TP-Link WR1043ND v1.8, OpenWRT 18.06.7

Description:

To reproduce: start a browser that will restore a saved session, performing a number of look-ups.

CPU usage will skyrocket. On occasion, memory use will also surge triggering an OoM crash that results in a zombie that slows down the router to a crawl (load reads 2 to 3)

Initially, I thought it was just the router having too little memory, however, after switching to stubby, which appears to have a similar memory footprint, everything has been working flawlessly.

stangri, 2020-02-23:
@aarond10.

jamesbo13, 2020-04-09:
Note - there was a recent patch in upstream to fix a memory leak (https://github.com/aarond10/https_dns_proxy/commit/40647ce94c62a47e9d53efae8018fb3142e277b9) ... perhaps this might be related

stangri, 2020-04-11:
Thanks, PRs for the new version: https://github.com/openwrt/packages/pull/11819, https://github.com/openwrt/packages/pull/11820, https://github.com/openwrt/packages/pull/11821.
Once they are merged, the new package should be available in repos withing a few days, please test if it fixes the memory leak.

dmascord, 2020-05-27:
This seems potentially related to the ipv4 vs ipv6 configuration.

https://openwrt.org/docs/guide-user/services/dns/doh_dnsmasq_https-dns-proxy suggests to use ipv4_resolvers setting, but the init scripts use the opposite configuration use_ipv6_resolvers_only.

I have  modified my configuration to use use_ipv6_resolvers_only 0, rather than ipv4_resolvers 1, and will monitor to see how it behaves.

stangri, 2020-05-27:
@dmascord could you please post your full https-dns-proxy and dnsmasq configs?

dmascord, 2020-05-27:
`
root@tp4320:/etc/config# cat https-dns-proxy

config main 'config'
	option update_dnsmasq_config '*'

config https-dns-proxy
	option bootstrap_dns '1.1.1.1,1.0.0.1'
	option resolver_url 'https://cloudflare-dns.com/dns-query'
	option listen_addr '127.0.0.1'
	option listen_port '5053'
	option user 'nobody'
	option group 'nogroup'
	option use_ipv6_resolvers_only '0'
	option verbosity '1'

root@tp4320:/etc/config# cat dhcp

config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option nonegcache '0'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option nonwildcard '1'
	option localservice '1'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option serversfile '/tmp/adb_list.overall'
	option confdir '/tmp/dnsmasq.d'
	option localuse '1'
	list server '127.0.0.1#5053'
	list server '0::1#5053'
	option noresolv '1'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv6 'disabled'
	list dhcp_option '6,10.0.0.1'
	option ra 'disabled'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
`

stangri, 2020-05-27:
Which version of https-dns-proxy is that?

dmascord, 2020-05-28:
strings /usr/sbin/https-dns-proxy:
**https-dns-proxy-2020-04-09**/src/main.c

stangri, 2020-05-28:
Did you modify the init script in any way?

willymilkytron, 2020-08-02:
Same issue here, https://github.com/aarond10/https_dns_proxy/commit/40647ce94c62a47e9d53efae8018fb3142e277b9 does not fix it for me.

I noticed memory usages usually increases when my ISP router reboots:

> network -> openwrt (https-dns-proxy) -> ISP router -> internet

aarond10, 2020-08-04:
Thanks @willymilkytron, that's a useful data point. There is at least one bad thing going on with libcurl but I haven't figured out exactly what yet. Worked fine for months, so I suspect it's triggered by either IP changes (DNS round robin?) or some http/2 server behavior.

stangri, 2020-08-04:
@aarond10 do you think it may be beneficial to test with `-x` on the systems with abnormal CPU/memory usage?

caliban511, 2020-09-27:
同样遇到 https_dns_proxy 占用CPU过高的问题。我的CPU是J4205，但这个进程一开，CPU平均占用就会上升到50%左右。这个问题不知道什么时候能解决一下？

aarond10, 2020-09-28:
I'll need more details. Ideally some gprof output with a debug build or at
the least the output of the daemon running with -vvvv so I can try to infer
what it's doing.

caliban511, 2020-09-29:
你好，请看下最上方占用25%CPU的那个，就这一条就占用掉25%，然后整个Openwrt的占用从7%左右上升到45%以上。
> 
> I'll need more details. Ideally some gprof output with a debug build or at the least the output of the daemon running with -vvvv so I can try to infer what it's doing.

[image]

caliban511, 2020-09-29:
请问这两个任务是在做什么呢？为什么占用那么大？
> 
> I'll need more details. Ideally some gprof output with a debug build or at the least the output of the daemon running with -vvvv so I can try to infer what it's doing.
[image]

caliban511, 2020-09-29:
> 
> I'll need more details. Ideally some gprof output with a debug build or at the least the output of the daemon running with -vvvv so I can try to infer what it's doing.

大佬你好，请问你用过passwall.ipk这个软件吗？我试过用最新版19.07.4的原版openwrt，安装passwall，会自动安装 https-dns-proxy ，然后只要它被装上就会立即引发这个问题，即使没有启用passwall也一样。

caliban511, 2020-10-07:
我自己尝试编译lienol大佬的代码生成固件后，发现并没有这个问题。所以这应该是软件和原生固件之间的兼容性问题了。问题终结。

aarond10, 2020-10-07:
It's an odd compatibility issue. 25% I assume is one of four cores so it might have been spinning trying to do something. I'll close for now, but feel free to reopen it if you think it's still a problem.

aarond10, 2020-10-07:
On second thought, I'll leave this open for a little longer given the multiple reports. I've not been able to reproduce myself since the commit about a month ago now that forces curl to abort active HTTP2 sessions if the IP address of the resolver changes. I'm wondering if that was part of the problem here.

Mrnofish, 2020-10-07:
Isn't there a rule that all submissions should be in English?

If there is, it should be enforced. If not, there should be one.

caliban511, 2020-10-08:
Sorry, I don’t understand English. This is using Google Translate to reply. There are some unsatisfactory words. Please forgive me.
This problem of abnormal CPU usage occurred when I installed a software called passwall with the original openwrt firmware.
The source code of this software is here:
https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-passwall
When I try to use the source code modified by the same person, compile it into openwrt firmware and use this software, the above problem does not appear. The revised firmware source code is here:
https://github.com/Lienol/openwrt
Because I don't understand these codes and only use them, I can't make any constructive comments. It can only be simply considered that this may be a compatibility issue between the original firmware and the software. I can't solve it, I can only avoid it.
If this troubles you, I'm really sorry.
Thank you for your continued work and dedication.

> 
> Isn't there a rule that all submissions should be in English?
> 
> If there is, it should be enforced. If not, there should be one.

caliban511, 2020-10-08:
Sorry, I don’t understand English. This is using Google Translate to reply. There are some unsatisfactory words. Please forgive me.
This problem of abnormal CPU usage occurred when I installed a software called passwall with the original openwrt firmware.
The source code of this software is here:
https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-passwall
When I try to use the source code modified by the same person, compile it into openwrt firmware and use this software, the above problem does not appear. The revised firmware source code is here:
https://github.com/Lienol/openwrt
Because I don't understand these codes and only use them, I can't make any constructive comments. It can only be simply considered that this may be a compatibility issue between the original firmware and the software. I can't solve it, I can only avoid it.
If this troubles you, I'm really sorry.
Thank you for your continued work and dedication.

> On second thought, I'll leave this open for a little longer given the multiple reports. I've not been able to reproduce myself since the commit about a month ago now that forces curl to abort active [...]

obhasqr, 2020-10-16:
I found that setting up an unreachable doh server will cause high CPU usage.
I can‘t access Google doh server. If I set up Google doh server in https-dns-proxy, it will cause my router's CPU to be occupied by more than 50%.

Add Google doh:
[image]
Delete Google doh:
[image]

stangri, 2020-10-16:
@aarond10 does the comment above shed any light on possible fix for high CPU utilization?

stangri, 2020-12-13:
@caliban511 @obhasqr -- can you please test the `2020-08-21-1` version and see if you can still reproduce the problem.

obhasqr, 2020-12-14:
Hi @stangri , ```2020-08-21-1``` can still reproduce this issue.

nirui, 2020-12-25:
Ping https://github.com/aarond10/https_dns_proxy/issues/95

aarond10, 2021-01-17:
Sorry for the slowness. I just removed the libcurl workaround code. It shouldn't be needed and looks like it was actually the cause of this.

obhasqr, 2021-02-10:
`2021-01-17-1` works fine for me.

stangri, 2021-02-10:
> `2021-01-17-1` works fine for me.

Safe to close this then? Anyone can still reproduce the high cpu usage with the 2021 builds?

bjmi, 2021-02-12:
Both [Extended test](https://www.dnsleaktest.com/) and [DNS Spoofability Test](https://www.grc.com/dns/dns.htm) don't let https-dns-proxy crash or stay at 100% CPU time.
```
opkg list-installed | grep https-dns-proxy
https-dns-proxy - 2021-01-17-1
```

NagaseKouichi, 2021-02-25:
Hi, I'm using https-dns-proxy - 2021-01-17-2 version, and it still have the out of memory problem. Here is the dmesg log:
[dmesg log.txt](https://github.com/openwrt/packages/files/6040887/dmesg.log.txt)

stangri, 2021-02-26:
> Hi, I'm using https-dns-proxy - 2021-01-17-2 version, and it still have the out of memory problem. Here is the dmesg log:
> [dmesg log.txt](https://github.com/openwrt/packages/files/6040887/dmesg.log.txt)

To be fair, before OOM in https-dns-proxy the attached dmesg shows OOM messages for haproxy and node/netdata. Am I reading it wrong or it doesn't look like it's a https-dns-proxy issue?

NagaseKouichi, 2021-02-26:
> To be fair, before OOM in https-dns-proxy the attached dmesg shows OOM messages for haproxy and node/netdata. Am I reading it wrong or it doesn't look like it's a https-dns-proxy issue?

Yes, but https-dns-proxy used the most memory. If I don't enable https-dns-proxy, system won't kill other applications.

stangri, 2021-02-28:
> Yes, but https-dns-proxy used the most memory. If I don't enable https-dns-proxy, system won't kill other applications.

You may want to share the package settings, to help reproduce the issue.

willymilkytron, 2021-03-26:
The memory leak is still happening on my router running 19.07.5 with 

```
https-dns-proxy - 2021-01-17-3
luci-app-https-dns-proxy - 2021-01-17-2
```

I'm trying to reproduce the issue on a different device since i can't install/use gdb and valgrind on the production router.

nirui, 2021-04-12:
Yep. I'm still encountering the unrecoverable 100% CPU usage problem, randomly. In fact, it happened so random, I can't reproduce it this time. 

A weird thing is, the `https-dns-proxy` does go to 100% CPU during "normal" situations on my router too, then it just back down after about a half minute. While during the "not-normal" situation, it does not go back down. My knowledge tells me, HTTPS requests should not take that amount of CPU time since it's mainly an I/O work of the network not CPU, unless something goes really wrong.

I'm currently running it in debug, hope something show up in the log.

So far, however, in addition to the CPU usage problem, I've also observed that the process `https-dns-proxy` has established a huge amount of connections to the HTTPS DNS server:

    # netstat -utpWn | grep ":443.*ESTABLISHED.*https-dns-"
    tcp     3184      0 10.154.148.133:43664                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:40244                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp     3184      0 10.154.148.133:43672                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:59674                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp     3184      0 10.154.148.133:59668                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:41528                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:54180                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:55538                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp       51      0 10.154.148.133:43958                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp     3184      0 10.154.148.133:36098                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:34084                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:38560                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp        0      0 10.154.148.133:34376                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp     3184      0 10.154.148.133:59962                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    tcp      275      0 10.154.148.133:60104                                223.5.5.5:443                                       ESTABLISHED 19107/https-dns-pro
    .... about 1000 lines total

There are many more `CLOSE_WAIT`s of course.

In the "System Log":

    Mon Apr 12 02:39:33 2021 daemon.info https-dns-proxy[19105]: [W] 1618195173.795973 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:35 2021 daemon.info https-dns-proxy[19107]: [W] 1618195175.875970 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:38 2021 daemon.info https-dns-proxy[19105]: [W] 1618195178.795946 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:40 2021 daemon.info https-dns-proxy[19107]: [W] 1618195180.866054 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:43 2021 daemon.info https-dns-proxy[19105]: [W] 1618195183.795978 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:45 2021 daemon.info https-dns-proxy[19107]: [W] 1618195185.875970 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:48 2021 daemon.info https-dns-proxy[19105]: [W] 1618195188.795930 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:50 2021 daemon.info https-dns-proxy[19107]: [W] 1618195190.875970 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:53 2021 daemon.info https-dns-proxy[19105]: [W] 1618195193.795942 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:55 2021 daemon.info https-dns-proxy[19107]: [W] 1618195195.875965 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:39:58 2021 daemon.info https-dns-proxy[19105]: [W] 1618195198.795958 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:40:00 2021 daemon.info https-dns-proxy[19107]: [W] 1618195200.875983 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
    Mon Apr 12 02:40:03 2021 daemon.info https-dns-proxy[19105]: [W] 1618195203.795932 src/dns_poller.c:66 DNS lookup failed: Could not contact DNS servers
 
Not sure if the two problem were related.

I'm running `OpenWrt 19.07.5 r11257-5090152ae3`,

    # curl -V
[... cut ...]
    
    # opkg info libcares
    Package: libcares
    Version: 1.15.0-4
    Depends: libc
    Status: install ok installed
    Section: libs
    Architecture: mipsel_24kc
    Size: 28012
    Filename: libcares_1.15.0-4_mipsel_24kc.ipk
    Description: c-ares is a C library for asynchronous DNS requests (including name resolves)
     
     C89 compatibility, MIT licensed, builds for and runs on POSIX, Windows,
     Netware, Android and many more operating systems.
    Installed-Time: 1612446180

stangri, 2021-04-12:
Have you tried using it with Cloudflare/Google servers and see if you can reproduce the error? Maybe it's something in the https-dns-proxy config or your resolver.

nirui, 2021-04-12:
Well, I have to use some "weird" configuration (a socks5h proxy on `127.0.0.1:1080`) for Google and Cloudflare server as they are not directly accessible in my network, so I expect their result to be even weirder.

Anyway, here is the result of `netstat -utpWn | grep "https-dns-" > /tmp/netstat.txt` (without filtering): [netstat.txt](https://github.com/openwrt/packages/files/6293781/netstat.txt). Notice there are many connections to `127.0.0.1:1080` too.

And here's the configuration of `https-dns-proxy` generated via `luci-app-https-dns-proxy`:

    # cat /etc/config/https-dns-proxy 

    config main 'config'
        option update_dnsmasq_config '*'
        list force_dns_port '53'
        list force_dns_port '853'
        option force_dns '0'

    config https-dns-proxy
        option resolver_url 'https://cloudflare-dns.com/dns-query'
        option listen_addr '127.0.0.1'
        option user 'nobody'
        option group 'nogroup'
        option bootstrap_dns '1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'
        option listen_port '5053'
        option verbosity '4'
        option proxy_server 'socks5h://127.0.0.1:1080'

    config https-dns-proxy
        option bootstrap_dns '8.8.8.8,8.8.4.4'
        option resolver_url 'https://dns.google/dns-query'
        option listen_addr '127.0.0.1'
        option user 'nobody'
        option group 'nogroup'
        option listen_port '5054'
        option verbosity '4'
        option proxy_server 'socks5h://127.0.0.1:1080'

    config https-dns-proxy
        option bootstrap_dns '208.67.222.222,208.67.220.220'
        option resolver_url 'https://doh.opendns.com/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5055'
        option verbosity '4'
        option proxy_server 'socks5h://127.0.0.1:1080'

    config https-dns-proxy
        option bootstrap_dns '116.202.176.26'
        option resolver_url 'https://doh.libredns.gr/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5056'
        option verbosity '4'
        option proxy_server 'socks5h://127.0.0.1:1080'

    config https-dns-proxy
        option bootstrap_dns '9.9.9.9,149.112.112.9,2620:fe::fe,2620:fe::9'
        option resolver_url 'https://dns9.quad9.net/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5057'
        option verbosity '4'
        option proxy_server 'socks5h://127.0.0.1:1080'

    config https-dns-proxy
        option bootstrap_dns '119.29.29.29,119.28.28.28'
        option resolver_url 'https://doh.pub/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5058'
        option verbosity '4'

    config https-dns-proxy
        option bootstrap_dns '223.5.5.5,223.6.6.6,2400:3200::1,2400:3200:baba::1'
        option resolver_url 'https://dns.alidns.com/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5059'
        option verbosity '4'

I've already tested the socks5 proxy. It works fine and no looping DNS query issues there.

aarond10, 2021-04-16:
Could you try removing the IPv6 bootstrap_dns, or set them to a consistent
and contactable server?
The DNS error you posted suggests that the bootstrap DNS resolution is
failing (dns_poller.cc, line 66). That shouldn't happen intermittently
unless you have a very very flaky network. More likely some of the DNS
server entries aren't working.

aarond10, 2021-04-16:
You can use `dig www.google.com @119.29.29.29` or similar to test. FWIW,
you only need those DNS servers to work to look up the main resolver
address so there isn't a real advantage to them all being different either.

nirui, 2021-05-28:
Here is the log generated during one of the event where the CPU usage spike:

    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.629568 main.c:92 Received request for id: b7ca, len: 34
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.629659 src/https_client.c:80 Requesting HTTP/1.1: 0
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]:
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.629708 src/https_client.c:118 Using curl proxy: socks5h://127.0.0.1:1080
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.641292 src/https_client.c:56 curl closed socket: 29
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.642224 src/https_client.c:32 curl opened socket: 34
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.643074 src/https_client.c:32 curl opened socket: 35
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.644156 src/https_client.c:56 curl closed socket: 34
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.644291 src/https_client.c:142 CURLINFO_EFFECTIVE_URL: https://dns.google/dns-query
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [D] 1622181741.644312 src/https_client.c:154 CURLINFO_RESPONSE_CODE: 0
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5492]: [E] 1622181741.644329 src/https_client.c:166 CURLINFO_OS_ERRNO: 146
    Fri May 28 06:02:21 2021 daemon.err https-dns-proxy[5494]: * Operation timed out after 10373 milliseconds with 0 out of 0 bytes received
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181738.531115 src/https_client.c:219 Times: 0.000379, 0.013635, 0.000000, 0.000000, 0.000000, 11.196436
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181738.531302 main.c:75 buflen 0
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]:
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.701196 main.c:92 Received request for id: aece, len: 32
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.701284 src/https_client.c:80 Requesting HTTP/1.1: 0
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]:
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.701331 src/https_client.c:118 Using curl proxy: socks5h://127.0.0.1:1080
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.704874 src/https_client.c:56 curl closed socket: 27
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.705541 src/https_client.c:32 curl opened socket: 32
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.706431 src/https_client.c:32 curl opened socket: 33
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.707422 src/https_client.c:56 curl closed socket: 32
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.707540 src/https_client.c:142 CURLINFO_EFFECTIVE_URL: https://cloudflare-dns.com/dns-query
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [D] 1622181741.707568 src/https_client.c:154 CURLINFO_RESPONSE_CODE: 0
    Fri May 28 06:02:21 2021 daemon.info https-dns-proxy[5494]: [E] 1622181741.707591 src/https_client.c:166 CURLINFO_OS_ERRNO: 146
    Fri May 28 06:02:22 2021 daemon.err https-dns-proxy[5495]: * Operation timed out after 10008 milliseconds with 0 out of 0 bytes received
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181738.797135 src/https_client.c:219 Times: 0.000404, 0.016502, 0.000000, 0.000000, 0.000000, 10.237343
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181738.797302 main.c:75 buflen 0
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]:
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.011176 main.c:92 Received request for id: b7ca, len: 34
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.011264 src/https_client.c:80 Requesting HTTP/1.1: 0
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]:
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.011309 src/https_client.c:118 Using curl proxy: socks5h://127.0.0.1:1080
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.014786 src/https_client.c:56 curl closed socket: 29
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.015438 src/https_client.c:32 curl opened socket: 34
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.016249 src/https_client.c:32 curl opened socket: 35
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.017209 src/https_client.c:56 curl closed socket: 34
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.017321 src/https_client.c:142 CURLINFO_EFFECTIVE_URL: https://doh.libredns.gr/ads
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [D] 1622181742.017344 src/https_client.c:154 CURLINFO_RESPONSE_CODE: 0
    Fri May 28 06:02:22 2021 daemon.info https-dns-proxy[5495]: [E] 1622181742.017366 src/https_client.c:166 CURLINFO_OS_ERRNO: 146
[... cut ...]
During the same time, the socks5h proxy (`127.0.0.1:1080`) was also generating multiple `getpeername: Socket not connected` errors, indicating it cannot connect to the remote. I guess there is a bug in `https-dns-proxy` that only randomly triggers during connection failures. 

That's all the information I've gotten so far. Hope it helps :)

And, yes, I guess my network is indeed flaky. I've asked my ISP to give me a network of a better tier, they agreed and are corrently installing this ASR 9000 router in my kitchen. I'll ask them to test the software again on this stabler network as soon as they figured how to flash OpenWrt on the router ;)

c0ke543, 2021-07-10:
This occurs for me when I add a firewall rules to block port 53 instead of hijacking port 53. Hopefully, this helps reporduce for people. I disabled the port 53 blocks and the issue went away. Here's the fire wall rules used.

I noticed port 53 was still opening when running namebench. Namebench won't run while hijacking. The rules below are disabled. 

config rule
	option target 'REJECT'
	option dest 'wan'
	option dest_port '53'
	option name 'Block port 53 leaks'
	option enabled '0'

config rule
	option dest_port '53'
	option src 'lan'
	option name 'Block port 53 leaks'
	option dest 'wan'
	option target 'REJECT'
	option enabled '0'

Here is a screen grab. I disabled then re-enabled. As you can see one I do a dns query. CPU get stuck in a loop with firewall rules in place. 

https://imgur.com/WIKoMEA

stangri, 2021-08-18:
I believe this has been resolved by https://github.com/openwrt/packages/pull/16252 please reopen if I'm wrong.

closed 2021-08-18 by stangri