openwrt/packages #11415: lua-libmodbus: demo-rtu.lua fails with Symbol not found: luaopen_libmodbus

hutorny, 2020-02-23:
Maintainer: @karlp  
Environment:  Atheros AR9330 rev 1 Carambola2, OpenWrt 19.07.1 ar71xx/generic, Lua 5.1.5

Description:
demo-rtu.lua, demo.lua and a trivial test
```
mb = require("libmodbus")
```
fails with error 
```
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> mb = require("libmodbus")
error loading module 'libmodbus' from file './libmodbus.so':
        Symbol not found: luaopen_libmodbus
stack traceback:
        [C]: ?
        [C]: in function 'require'
        stdin:1: in main chunk
        [C]: ?
>
```

suspecting missing `__attribute__ ((visibility ("default")))`
near [lua-libmodbus.c:1220](https://github.com/etactica/lua-libmodbus/blob/master/lua-libmodbus.c#L1220)

karlp, 2020-02-23:
Have you got any extra build hardening or something enabled?  this has always worked in the past, and "default" visibility doesn't sound like it should change anything?  Here's my own board, running a 19.07 based build, based on a carambola2 module as well:
```
root@eg-0366BD:~# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> mb = require("libmodbus")
> 
```
"just works"

hutorny, 2020-02-23:
It was a fresh OpenWRT installation from `openwrt-19.07.1-ar71xx-generic-carambola2-squashfs-sysupgrade.bin` + `libmodbus lua-libmodbus` on two  instances of carambola2.
Strangely enough, I cannot reproduce it myself any more on any of my carambolas, and I am not sure what was the remedy.

closed 2020-02-23 by hutorny