openwrt/packages #11417: samba4: the fix for building with parallel jobs was removed

dbermond, 2020-02-23:
**Maintainer:** @Andy2244

**Environment:**
- **Architecture:** arm_cortex-a15_neon-vfpv4
- **Model:** R7800
- **OpenWrt and packages:** git master

samba4 got a fix for building with parallel jobs at commit 710700c7b4e3890b309a1b8d219e9317d07cb7d5. The problem was reported at issue #8721.

But this fix was reverted/removed at commit c50c5836ea9b5b955d224b721208c2ee29bc9f62. As a result, the same problem reported at #8721 is now valid again: samba4 fails to build with parallel jobs enabled, giving the error: `/bin/sh: rpcgen: command not found`.

`rpcgen` executable (now provided by package [rpcsvc-proto)](https://github.com/openwrt/packages/tree/master/libs/rpcsvc-proto) needs to be at `HOST_BUILD_DEPENDS`. Applying this change fixes the issue for me and makes samba4 to build fine again with parallel jobs:
```patch
diff -Naurp a/net/samba4/Makefile b/net/samba4/Makefile
--- a/net/samba4/Makefile	2020-02-22 23:33:28.000000000 +0000
+++ b/net/samba4/Makefile	2020-02-23 00:37:47.091918977 +0000
@@ -21,8 +21,8 @@ PKG_LICENSE_FILES:=COPYING
 PKG_CPE_ID:=cpe:/a:samba:samba
 
 # samba4=(asn1_compile,compile_et) rpcsvc-proto=(rpcgen)
-HOST_BUILD_DEPENDS:=python3/host perl/host
-PKG_BUILD_DEPENDS:=samba4/host libtasn1/host rpcsvc-proto/host
+HOST_BUILD_DEPENDS:=python3/host perl/host rpcsvc-proto/host
+PKG_BUILD_DEPENDS:=samba4/host libtasn1/host
 
 PKG_CONFIG_DEPENDS:= \
 	CONFIG_SAMBA4_SERVER_NETBIOS \
```

If this change is acceptable I can open a pull request.

Andy2244, 2020-02-23:
Sure open a PR.

PS: I always build with `make -j8` on debian9 and never had this issue, so not sure why this actually happens.

dbermond, 2020-02-23:
Thank you for the reply :)

I've opened a pull request.

Maybe `rpcgen` is being installed by default or being pulled by other package in your Debian host? I'm building on Arch Linux and `rpcgen` is not pulled by the minimal OpenWrt build dependencies.

PR #11419 (2020-02-23): samba4: bring back the fix to build with parallel jobs
**Maintainer:**  @Andy2244
**Compile tested:** arm_cortex-a15_neon-vfpv4 (R7800), OpenWrt git master r12290
**Run tested:** arm_cortex-a15_neon-vfpv4 (R7800), OpenWrt git master r12290

Description:

Fixes #11417

The fix was implemented in commit 710700c and was later removed in commit c50c583.
commit b1aed107 (2020-02-23): samba4: bring back the fix to build with parallel jobs

Fixes #11417

The fix was implemented in commit 710700c and was later removed in
commit c50c583.
closed 2020-02-23 by neheb