openwrt/packages #11431: mwan3: After failure and recovery of primary interface, backup ARPs for Internet hosts

brianjmurrell, 2020-02-25:
Maintainer: @feckert 
Environment: 19.07.1 on QEMU x86_64

Description:
I have two wan interfaces, `wan` and `wanb` with a general primary/backup policy.

I start a ping from the router to an host on the Internet and observe that those are successfully using the primary interface.  When the primary fails, the pings switch to the backup interface successful.

However when the primary comes back, the secondary then goes into failure mode because rather than routing the tracking pings it wants to do to the upstream router, it starts ARPing for those addresses, as if it believes they are on the local network.  I.e.

```
11:25:41.787681 ARP, Request who-has 208.67.220.220 tell 192.168.101.249, length 46
11:25:42.827577 ARP, Request who-has 208.67.220.220 tell 192.168.101.249, length 46
11:25:51.749125 ARP, Request who-has 208.67.222.222 tell 192.168.101.249, length 46
11:25:52.827538 ARP, Request who-has 208.67.222.222 tell 192.168.101.249, length 46
11:25:53.750816 ARP, Request who-has 208.67.220.220 tell 192.168.101.249, length 46
```

And of course there is now no response from the tracking pings.

Prior to the primary going down and coming back up, the backup interface properly routes those tracking pings an thus gets responses from them.  This switching to ARPing for them only happens after the primary interface has come back and the ARPing will not stop until I reset the backup interface with an `ifdown wanb; ifup wanb`.

My `/etc/config/mwan3`:

```
config rule 'SMTP'
	option dest_port '25'
	option proto 'tcp'
	option sticky '0'
	option use_policy 'wanb_only'

config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan_wanb'

config globals 'globals'
	option mmx_mask '0x3F00'
	option rtmon_interval '5'

config interface 'wan'
	option enabled '1'
	list track_ip '8.8.4.4'
	list track_ip '8.8.8.8'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option reliability '2'
	option count '1'
	option timeout '2'
	option failure_latency '1000'
	option recovery_latency '500'
	option failure_loss '20'
	option recovery_loss '5'
	option interval '5'
	option down '3'
	option up '8'

config interface 'wan6'
	option enabled '0'
	list track_ip '2001:4860:4860::8844'
	list track_ip '2001:4860:4860::8888'
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'
	option family 'ipv6'
	option reliability '2'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'

config interface 'wanb'
	list track_ip '8.8.4.4'
	list track_ip '8.8.8.8'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option reliability '1'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	option enabled '1'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option failure_interval '5'
	option recovery_interval '5'

config interface 'wanb6'
	option enabled '0'
	list track_ip '2001:4860:4860::8844'
	list track_ip '2001:4860:4860::8888'
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'
	option family 'ipv6'
	option reliability '1'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'

config member 'wan_m1_w3'
	option interface 'wan'
	option metric '1'
	option weight '3'

config member 'wan_m2_w3'
	option interface 'wan'
	option metric '2'
	option weight '3'

config member 'wanb_m1_w2'
	option interface 'wanb'
	option metric '1'
	option weight '2'

config member 'wanb_m2_w2'
	option interface 'wanb'
	option metric '2'
	option weight '2'

config member 'wan6_m1_w3'
	option interface 'wan6'
	option metric '1'
	option weight '3'

config member 'wan6_m2_w3'
	option interface 'wan6'
	option metric '2'
	option weight '3'

config member 'wanb6_m1_w2'
	option interface 'wanb6'
	option metric '1'
	option weight '2'

config member 'wanb6_m2_w2'
	option interface 'wanb6'
	option metric '2'
	option weight '2'

config policy 'wan_only'
	list use_member 'wan_m1_w3'
	list use_member 'wan6_m1_w3'

config policy 'wanb_only'
	list use_member 'wanb_m1_w2'
	list use_member 'wanb6_m1_w2'

config policy 'balanced'
	list use_member 'wan_m1_w3'
	list use_member 'wanb_m1_w2'
	list use_member 'wan6_m1_w3'
	list use_member 'wanb6_m1_w2'

config policy 'wan_wanb'
	list use_member 'wan_m1_w3'
	list use_member 'wanb_m2_w2'
	list use_member 'wan6_m1_w3'
	list use_member 'wanb6_m2_w2'

config policy 'wanb_wan'
	list use_member 'wan_m2_w3'
	list use_member 'wanb_m1_w2'
	list use_member 'wan6_m2_w3'
	list use_member 'wanb6_m1_w2'
```

brianjmurrell, 2020-02-26:
What's interesting here is that this ARPing for non-local addresses only happens on the backup interface once the primary has recovered.  The tracking pings on the primary interface DO NOT ARP for their targets, they route them upstream as they should be.

I find it entirely odd that on a given host, one interface will know remote hosts are not local and route packets for them and a different interface will be confused about what is local and will try to ARP request for all IPs it tries to reach, both local and remote.

feckert, 2020-02-26:
Have you set any routind metric to the interfaces? I can not see them in your output?

brianjmurrell, 2020-02-26:
I have not knowingly set any routing metric to the interfaces.  What command output can I provide you to confirm?

brianjmurrell, 2020-02-27:
Happy to provide whatever output is needed to verify if any routing metric has been set to the interfaces.

feckert, 2020-02-28:
See https://github.com/openwrt/packages/issues/11440#issuecomment-592527021

brianjmurrell, 2020-02-28:
Yes, that was it.  I missed out on Step 1 of https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#prepare_and_the_check_the_default_routing_table_for_the_wan_interfaces.

Apologies for the noise.

closed 2020-02-28 by brianjmurrell

brianjmurrell, 2020-02-29:
It seems that probably mwan3 could warn and refuse to start if these route metrics are not configured.

Or is there any reason mwan3 cannot just assign metrics?  Are the metrics actually used in any way while mwan3 is running or does mwan3's routing tables end up overriding the default routing table and it's metrics?

feckert, 2020-03-07:
> It seems that probably mwan3 could warn and refuse to start if these route metrics are not configured.

The would be great. Feel free to open a pullrequest for this change.

> Or is there any reason mwan3 cannot just assign metrics? Are the metrics actually used in any way while mwan3 is running or does mwan3's routing tables end up overriding the default routing table and it's metrics?

I think this is also possible as an option to add a random metric to this interface if they are not already set. The main thing is that the metrics are different.