openwrt/packages #11441: mwan3: restart renders router unconnectable, wont route packackets anymore until restart

misanthropos, 2020-02-27:
Maintainer: feckert@\<github-user> 
Environment: (ath79, Archer C7v2, 19.07.1)

Description:
Any change when using commit+save on mwan3 will need a restart of the router, because no
traffic is answered anymore.
I tried to figure out with logread what is going on, but the last  message is restarting mwan3

My setup has 4 OpenVPN connections, I supplied my config here:
[mwan3.txt](https://github.com/openwrt/packages/files/4258823/mwan3.txt)

neheb, 2020-02-27:
ping @feckert

feckert, 2020-02-28:
I do not use mwan3 for openvpn. My use case is **always** use mwan3 for physical interfaces. And not for virtual interfaces.

jamesmacwhite, 2020-05-04:
@misanthropos I have observed you may need to restart OpenVPN with `/etc/init.d/openvpn restart` when mwan3 is restarted _sometimes_. What could be happening is when mwan3 is restarted the VPN interfaces could be marked as down and that stops the traffic from being routed, but restarting the OpenVPN client process themselves should bring them back. Having to power cycle your router shouldn't be needed.

Have you observed the interface status of your VPN interfaces from the mwan3 status page when this happens?

feckert, 2020-05-04:
You could also add this to mwan3.user script which is called on events

jamesmacwhite, 2020-05-04:
@feckert Good point, I think that's worth doing. I might implement that myself.

misanthropos, 2020-05-04:
I forgot about this issue - this happens in certain configurations but I do not know which ones. With the "wrong" config it will fail every time mwan3 restarts.

jamesmacwhite, 2020-05-04:
@misanthropos When the connectivity fails what is observed when running `mwan3 status` or looking at the interface status in LuCI?

misanthropos, 2020-05-04:
I cannot tell - there is NO connectivity to the router left - I would have to connect a serial console which I do not have to see what is happening.

jamesmacwhite, 2020-05-04:
Sorry for misunderstanding, I thought you meant VPN clients didn't work with mwan3 active, not the whole router. That seems very strange. That's certainly not normal behaviour. Are you running any other policy routing based applications with mwan3 by any chance?

I've not known mwan3 to break LAN based routing like that.

Are all your VPN interfaces having the default gateway prevented with `route-nopull` or similar?

misanthropos, 2020-05-04:
yes - all VPN configs have route-nopull

```
pull-filter ignore redirect-gateway
route-nopull
route 0.0.0.0 128.0.0.0 vpn_gateway XX 
```
XX is a different metric for every VPN gatway.

the last resort option would not stop the lan ports of being inactive (unrouted?) either.

jamesmacwhite, 2020-05-04:
@misanthropos I think you don't need to have all of the combinations relating to overriding the routes, for OpenVPN 2.4, `pull filter ignore redirect-gateway` should be enough on it's own. 

I know the Wiki recommends adding `route 0.0.0.0 0.0.0.0 vpn_gateway 20` to assign a metric, but technically it can work without this, it's just best practice to have some form of metric value. I have configured a load balanced VPN policy without metrics. You get away with it because it's not the default route and a specific policy is stating to use the VPN interfaces through mwan3.

I'm not sure about that route though:

```route 0.0.0.0 128.0.0.0 vpn_gateway XX```

If the purpose is to prevent the default gateway being overridden, you don't need this for it. Can you clarify what this is for?

No, unreachable shouldn't but `default` means fallback to the main routing table, which is often best for a WAN interface.

VPN bypass shouldn't in theory conflict, mark sure the fwmark values being used don't collide with mwan3, I think stangri did make sure they don't, but good to check. Possibly test with vpnbypass disabled with mwan3 only active.

That's all the assistance I can offer based on what you've described, so sorry if it doesn't help.

misanthropos, 2020-05-04:
Thank you very much for your input.

Regarding the route entry:
It just masks the highest bit but I do not know why that is.
What I did was just just to "emulate" the default behaviour of my VPN - thats the gw mask if I take over their settings. Could be related to protect the default gw - but could also be the reason why some connections dont go through VPN.... which is a different topic

feckert, 2020-06-29:
@jamesmacwhite thanks for your work an help!
@misanthropos did you solve your issue. I want to apologize for not being able to contribute more to the problem solving.
Feel free to reopen the issue.

closed 2020-06-29 by feckert

misanthropos, 2020-06-30:
I have not - but I was not changing anything anymore - it happend with certain configurations and I dont know how to reproduce it. Only that it seems to be related to OpenVPNs managed by mwan3.