openwrt/packages #11457: iptables-mod-tproxy  is missing for 19.07.1

cutekevin7, 2020-02-28:
opkg install iptables-mod-tproxy
Unknown package 'iptables-mod-tproxy'.
Collected errors:
 * opkg_install_cmd: Cannot install package iptables-mod-tproxy.

and checked the latest version is for 19.07.0.  expect the updated version for the latest stable version of openwrt.

many thanks
K.

neheb, 2020-02-28:
Which platform?

Sounds bogus. Run opkg update first.

cutekevin7, 2020-03-08:
> Which platform?
> 
> Sounds bogus. Run opkg update first.

already opkg update.

please refer to [https://openwrt.org/packages/pkgdata/iptables-mod-tproxy](https://openwrt.org/packages/pkgdata/iptables-mod-tproxy)

it says 

> package: iptables-mod-tproxy
> Name: iptables-mod-tproxyVersion: 1.8.3-1
> Description: Transparent proxy iptables extensions.\\ \\ Matches: \\ - socket\\ \\ Targets: \\ - TPROXY\\ \\ \\
> Installed size: 3kB
> Dependencies: libc, iptables, kmod-ipt-tproxy
> Categories: network---firewall
> Repositories: base
> OpenWrt release: **OpenWrt-19.07.0**
> File size: 4kB
> License: GPL-2.0
> Maintainer: OpenWrt team

cutekevin7, 2020-03-08:
> Which platform?
> 
> Sounds bogus. Run opkg update first.

btw
aarch64_cortex-a53

neheb, 2020-03-08:
I see this: https://archive.openwrt.org/releases/19.07.0/targets/mvebu/cortexa53/kmods/4.14.162-1-ce57010438699985314b7e62055dc961/kmod-ipt-tproxy_4.14.162-1_aarch64_cortex-a53.ipk

neheb, 2020-03-08:
https://archive.openwrt.org/releases/19.07.0/targets/mvebu/cortexa53/packages/iptables-mod-tproxy_1.8.3-1_aarch64_cortex-a53.ipk

cutekevin7, 2020-03-14:
> https://archive.openwrt.org/releases/19.07.0/targets/mvebu/cortexa53/packages/iptables-mod-tproxy_1.8.3-1_aarch64_cortex-a53.ipk

thanks a lot.  done.

closed 2020-03-14 by cutekevin7

leosh64, 2020-04-27:
Same issue here - the package seems not to be included in **19.07.1**, but only in **19.07.0**. Manually installing the `.ipk` files is tedious due to the dependency chain: 
```
opkg install iptables-mod-tproxy_1.8.3-1_aarch64_cortex-a53.ipk
Installing iptables-mod-tproxy (1.8.3-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for iptables-mod-tproxy:
 *      kmod-ipt-tproxy
 * opkg_install_cmd: Cannot install package iptables-mod-tproxy.
```
Any progress on this issue?

neheb, 2020-04-30:
No. This belongs in https://bugs.openwrt.org

leosh64, 2020-10-18:
I've got it to work by force-installing `kmod-ipt-tproxy` even though the kernel version doesn't match:
```
# install kmod-ipt-tproxy and ignore dependency errors:
opkg install kmod-ipt-tproxy_4.14.162-1_aarch64_cortex-a53.ipk --force-depends
# install iptables-mod-tproxy
opkg install iptables-mod-tproxy_1.8.3-1_aarch64_cortex-a53.ipk
```
Now packages that depend on `iptables-mod-tproxy` (such as `shadowsocks-libev-ss-rules`) can be installed the regular way:
```
opkg install shadowsocks-libev-ss-rules
```
Hope this helps!