openwrt/packages #11461: Eror kmod-ath9k image builder openwrt snapshoot ath79

server46, 2020-02-29:
`     Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ath9k:
 * iw
 * opkg_install_cmd: Cannot install package kmod-ath9k.
Makefile:153: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:112: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:196: recipe for target 'image' failed
make: *** [image] Error 2   `

BKPepe, 2020-03-02:
This repository is for OpenWrt packages. This issue was already reported in OpenWrt mailing list:
https://lists.infradead.org/pipermail/openwrt-devel/2020-February/021928.html

And should be reported to the main OpenWrt repository.

closed 2020-03-02 by BKPepe

serverlinuxit, 2020-03-02:
> This repository is for OpenWrt packages. This issue was already reported in OpenWrt mailing list:
> https://lists.infradead.org/pipermail/openwrt-devel/2020-February/021928.html
> 
> And should be reported to the main OpenWrt repository.

Thanks