openwrt/packages #11470: minidlna init script reload broken

jow-, 2020-03-01:
Commit e9a65f13808cb4d9bd044de47ace5db008ebb117 refactored the init script of minidlna which broke config generation and service reloads in the process.

Issue 1)

The `procd_send_signal "$PROG"` invocation does not work since `$PROG` contains the absolute path to the executable while `procd_send_signal` expects a service name. This can be observed by running the script with `INIT_TRACE=1`:

```
root@jj:~# INIT_TRACE=1 /etc/init.d/minidlna reload
[...]
+ ubus call service signal '{ "name": "\/usr\/sbin\/minidlnad" }'
Command failed: Not found
[...]
```

Issue 2)

The init script registers `procd_add_reload_trigger "minidlna"` but fails to regenerate `/var/etc/minidlna.conf` on `/etc/init.d/minidlna reload`, breaking the established workflow for applying configuration changes.

cc @neheb

neheb, 2020-03-01:
Will fix when I can.

neheb, 2020-03-05:
Merged https://github.com/openwrt/packages/pull/11499

closed 2020-03-05 by neheb