openwrt/packages #11474: exfat-nofuse: Compilation failed on x86-64 kernel 5.4

DazzyWalkman, 2020-03-02:
Test compilation failed on current x86-64 snapshot with test kernel version 5.4 with error:
 > else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & MS_RDONLY)) {
 >                                                               ^~~~~~~~~
>                                                                IS_RDONLY                                                            

A patch like this may be needed:
https://github.com/dorimanx/exfat-nofuse/issues/145
Would someone more experienced have a look?
Thanks.

neheb, 2020-03-04:
Wonder if this package should be dropped.

Kernel 5.4 has an exfat driver in the staging directory.

DazzyWalkman, 2020-03-05:
I think so.  It only depends on how difficult it would be to backport the exfat staging driver to older kernels for OpenWrt.

neheb, 2020-03-05:
Next release is 5.4 only. So moot point.

closed 2020-03-09 by DazzyWalkman