openwrt/packages #11479: stubby:  fails to start with error "can't create : nonexistent directory"

jorti, 2020-03-02:
Maintainer: @jonathanunderwood 
Environment: x86_64, qemu VM, master branch

Description:
Stubby is failing to start since I updated to latest version in master some days ago. I have to set `option manual '1'` in `/etc/config/stubby` to be able to start it.

```
root@OpenWrt:~# /etc/init.d/stubby start
/etc/rc.common: line 21: can't create : nonexistent directory
mv: can't rename '/var/etc/stubby/stubby.yml.28731': No such file or directory
```

Enabling debug in the init script I can see:

```
+ config_get_bool manual global manual 0
+ local _tmp
+ config_get _tmp global manual 0
+ eval export -n -- '_tmp=${CONFIG_global_manual:-${4}}'
+ export -n -- '_tmp=0'
+ _tmp=0
+ export -n 'manual=0'
+ '[' 0 '=' 1 ]
+ config_file_tmp=/var/etc/stubby/stubby.yml.29925                                                                                                   
+ generate_config /var/etc/stubby/stubby.yml.29925
/etc/rc.common: line 22: can't create : nonexistent directory
+ mv /var/etc/stubby/stubby.yml.29925 /var/etc/stubby/stubby.yml
mv: can't rename '/var/etc/stubby/stubby.yml.29925': No such file or directory
```

anomeome, 2020-03-02:
I think this was addressed a few days back?

jorti, 2020-03-03:
You're right, it's fixed in latest master.

Thank you.

closed 2020-03-03 by jorti