openwrt/packages #11480: [OpenWrt 19.07] libaio package can not be compiled with enabled ASLR_PIE

BKPepe, 2020-03-02:
Maintainer: @sbyx 
Environment: Turris 1.1, powerpc_8540, OpenWrt 19.07.2

Description:

With explicitly enabled ASLR PIE, I am not able to compile libaio package.

```
/home/pepe/work/turrisos/turris1x-hbl/build/staging_dir/toolchain-powerpc_8540_gcc-7.5.0_musl/lib/gcc/powerpc-openwrt-linux-musl/7.5.0/../../../../powerpc-openwrt-linux-musl/bin/ld: io_queue_run.os: in function `io_queue_run':
io_queue_run.c:(.text+0x6c): undefined reference to `__stack_chk_fail_local'
/home/pepe/work/turrisos/turris1x-hbl/build/staging_dir/toolchain-powerpc_8540_gcc-7.5.0_musl/lib/gcc/powerpc-openwrt-linux-musl/7.5.0/../../../../powerpc-openwrt-linux-musl/bin/ld: compat-0_1.os: in function `compat0_1_io_cancel':
compat-0_1.c:(.text+0x48): undefined reference to `__stack_chk_fail_local'
/home/pepe/work/turrisos/turris1x-hbl/build/staging_dir/toolchain-powerpc_8540_gcc-7.5.0_musl/lib/gcc/powerpc-openwrt-linux-musl/7.5.0/../../../../powerpc-openwrt-linux-musl/bin/ld: compat-0_1.os: in function `compat0_1_io_queue_wait':
compat-0_1.c:(.text+0xcc): undefined reference to `__stack_chk_fail_local'
/home/pepe/work/turrisos/turris1x-hbl/build/staging_dir/toolchain-powerpc_8540_gcc-7.5.0_musl/lib/gcc/powerpc-openwrt-linux-musl/7.5.0/../../../../powerpc-openwrt-linux-musl/bin/ld: compat-0_1.os: in function `compat0_1_io_getevents':
compat-0_1.c:(.text+0x150): undefined reference to `__stack_chk_fail_local'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:53: libaio.so.1.0.1] Error 1

```

BKPepe, 2020-03-02:
Version 0.3.112 from `master` branch with removed patch `005_destdir.patch` fixes this issue for me. I will do a PR in a sec.

neheb, 2020-03-02:
Weird. That removed patch should be upstream. In any case, merged. Closing.

PR #11482 (2020-03-02): [OpenWrt 19.07] libaio: Update to 0.3.112
Maintainer: @neheb/@styx
Compile tested: Turris 1.1, powerpc_8540, OpenWrt 19.07.2

Description:
Fixes #11480

Cherry-picked commit from @neheb, which updates libaio to version 0.3.112
closed 2020-03-02 by neheb