openwrt/packages #11505: docker-ce: info command shows warning

feckert, 2020-03-04:
Maintainer: @G-M0N3Y-2503 
Environment: x86_64, APU3, OpenWrt master)

Description:

On executing the command ```docker info``` I get the following WARNING: on the output

```
WARNING: bridge-nf-call-iptables is disabled
WARNING: bridge-nf-call-ip6tables is disabled
```

The Problem is that this Options are disabled in https://github.com/openwrt/openwrt/blob/master/package/kernel/linux/files/sysctl-br-netfilter.conf
I have some problems with the macvlan driver on Openwrt and I think this is the root cause of my issue.

G-M0N3Y-2503, 2020-03-12:
It looks like creating a `/etc/sysctl.d/12-br-netfilter.conf` with
```conf
net.bridge.bridge-nf-call-ip6tables=1
net.bridge.bridge-nf-call-iptables=1
```
Stops the warning, would installing this as part of the package solve this issue?

feckert, 2020-03-12:
As it seems, the problem with macvlan is not this setting. As it looks, it is generally not possible for the container to reach its own host with OpenWrt. I used the following howto to setup macvlan https://blog.oddbit.com/post/2018-03-12-using-docker-macvlan-networks/ to ping the docker host from the container. 

Nevertheless the question is whether the warning is valid. Maybe it still affects docker.
So we could add a systcl.d file `12-docker-br-netfilter.conf`` to enable this setting and so remove the warning.

But first of all we should understand why docker needs this configuration or prints a warning.
@G-M0N3Y-2503

G-M0N3Y-2503, 2020-03-13:
Looking [at the code here](https://github.com/docker/docker-ce/search?q=BridgeNFCallIPTablesDisabled&unscoped_q=BridgeNFCallIPTablesDisabled) and [here](https://github.com/docker/docker-ce/search?l=Go&q=BridgeNfIptables) it seems that all it does is check if the options are enabled and prints said warning if not. It seems like docker wants this on because it [adds rules to the iptables](https://docs.docker.com/network/iptables/).
If the macvlan issue is related to this I would guess that there are conflicting IP tables rules added by docker.

feckert, 2020-03-13:
If docker already warns that the firewall is not set up properly, then I think we should also enable the option regardless of whether we have a problem or not.

Regardless of the setting, macvlan doesn't work on the router, so I can't access the Internet through the router from my container.  Can someone please confirm this?

lukasmrtvy, 2020-03-22:
@G-M0N3Y-2503 

Little hijack.. 
- Would be great to create package also for https://podman.io/ ( dockerd replacement )-> it can work almost 1:1 to docker-cli.  Whole community is moving from docker ( Acquisition from Mirantis ) and moby is lost in the space..
- Seems there is already some luci-app for docker -> https://github.com/lisaac/luci-app-dockerman

closed 2020-05-05 by feckert