openwrt/packages #11514: dbus,  Error: dbus[5175]: Unknown group "netdev" in message bus configuration file

TLCary, 2020-03-05:
Maintainer: @\common packages maintainers
Environment: (ARMv7, WRT3200ACM, 19.07)

Description:
`Error: dbus[5175]: Unknown group "netdev" in message bus configuration file`

New install of OpenWRT, I attempted to installed shairport-sync-openssl 3.2.2-3 thru LuCi.  I contacted the Maintainer of shairport who refereed me to the dbus maintainer.

Install Log from the Package Manager

```
Installing shairport-sync-openssl (3.2.2-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/shairport-sync-openssl_3.2.2-3_arm_cortex-a9_vfpv3.ipk
Installing libexpat (2.2.7-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/libexpat_2.2.7-1_arm_cortex-a9_vfpv3.ipk
Installing libdbus (1.12.12-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/libdbus_1.12.12-1_arm_cortex-a9_vfpv3.ipk
Installing dbus (1.12.12-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/dbus_1.12.12-1_arm_cortex-a9_vfpv3.ipk
Installing libavahi-dbus-support (0.7-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/libavahi-dbus-support_0.7-3_arm_cortex-a9_vfpv3.ipk
Installing avahi-dbus-daemon (0.7-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/avahi-dbus-daemon_0.7-3_arm_cortex-a9_vfpv3.ipk
Installing libavahi-client (0.7-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/libavahi-client_0.7-3_arm_cortex-a9_vfpv3.ipk
Installing libsoxr (0.1.3-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/arm_cortex-a9_vfpv3/packages/libsoxr_0.1.3-1_arm_cortex-a9_vfpv3.ipk
Configuring libexpat.
Configuring libdbus.
Configuring dbus.
Configuring libavahi-dbus-support.
Configuring avahi-dbus-daemon.
Configuring libavahi-client.
Configuring libsoxr.
Configuring shairport-sync-openssl.
```

neheb, 2020-03-06:
dbus is not maintained.

It's also quite out of date. I have a local patch that introduces dbus-broker, but I have not made much progress in getting it working.

neheb, 2020-03-06:
Seems shairport-sync-mini does not use dbus.

TLCary, 2020-03-06:
Homebridge does as well, and fails to work.  I keep hitting walls trying to install packages and it keeps coming back to this.  There are packages listed, and maintainers assigned... and they error and fail and no one is fixing them.  Passing the buck doesn't make a system work better.  If the packages are listed by default with an OpenWRT install, shouldn't they work?

closed 2020-04-25 by neheb