openwrt/packages #11521: The calculation of the memory limitation

HiGarfield, 2020-03-07:
https://github.com/openwrt/packages/blob/42af2583ad7ab757c2a2a738348c5606c3d27b77/net/transmission/files/transmission.init#L69

When we use **ulimit -a**, we can find that the unit is in kb, but it seems that byte is used here. Is it correct?

neheb, 2020-03-07:
Beats me. @nmav wrote it I believe.

I personally do not use it (I set the percentagt to 100).

HiGarfield, 2020-03-07:
> Beats me. @nmav wrote it I believe.
> 
> I personally do not use it (I set the percentagt to 100).

Setting the percentagt to 100 makes my router crash and reboot repeatedly.

neheb, 2020-03-07:
I'm guessing same thing happens with that routine removed.

HiGarfield, 2020-03-08:
> I'm guessing same thing happens with that routine removed.

Yes. When the memory allocation is not limited, the router will crash due to transmission. That is why I want to find a way to solve it.

By the way, I have tried to use **USE=$((MEM * mem_percentage / 1000))** . In this case, router will not crash, but transmission is not working normally. Though the system log tells me transmission has been started, I cannot open the transmission web interface.

neheb, 2020-03-08:
What router is this on?

HiGarfield, 2020-03-09:
> What router is this on?

DomyWifi DW33D with QCA9558 and 256M RAM

neheb, 2020-03-09:
Weird. Try adding a swap device.

HiGarfield, 2020-03-09:
> Weird. Try adding a swap device.

I have already added 2GB swap. Still crash.

Another interesting finding is that adding the memory limit in init script makes the router crash more frequently with transmission. So I am wondering whether the memory limitation is done in a correct way.

HiGarfield, 2020-03-10:
Another finding is that the frequently crash/reboot may be caused by a recently added torrent that includes an extremely large file sized more than 20GB. Transmission occupies more and more cache memory during downing, which can not be reclaimed by Linux kernel. This eventually causes the crash/reboot of the router. So memory limitation is still a big issue.

HiGarfield, 2020-03-10:
I have tried some other torrents and find that limiting the download speed has nothing to do with preventing/reducing the crash/reboot. As long as the files in the torrent are large enough (more than 15GB per file in my cases), transmission will continuously consume more and more cache memory (even though the download speed is limited to 200KB/s), until the router crashes and reboots itself.

neheb, 2020-03-17:
https://github.com/transmission/transmission/issues/313 says static linking prevents this. OTOH, it means larger binaries.

If you really want, I can provide test binaries for your platform.

neheb, 2020-03-17:
Oh that's disgusting. Reading that thread reveals the problem to be libcurl. Specifically non-openssl versions...

That would indicate upstream needs to provide multiple packages...

neheb, 2020-03-17:
Are you using transmission-openssl or transmission-mbedtls?

HiGarfield, 2020-03-24:
> transmission-openssl

transmission-openssl

HiGarfield, 2020-03-24:
> Are you using transmission-openssl or transmission-mbedtls?

transmission-openssl

HiGarfield, 2020-03-24:
> [transmission/transmission#313](https://github.com/transmission/transmission/issues/313) says static linking prevents this. OTOH, it means larger binaries.
> 
> If you really want, I can provide test binaries for your platform.
> 
> edit: OpenWrt also has rtorrent available. Could be worth a shot.

I have also read that issue. That problem is related to Debian, rather than openwrt. Moreover, that issue is on transmission 2.92. Maybe a different case in openwrt.

closed 2020-04-26 by HiGarfield