openwrt/packages #11531: node: cannot find libatomic.so.1.0.0

zokl, 2020-03-08:
Maintainer: @nxhack
Environment: (arm_cortex-a7+neon-vfpv4, Olimex A20 Micro, Master)

Description:

I am not able to compile the node as a package to the target system. Compilation fails on followed error:
```
/../staging_dir/hostpkg/share/icu/65.1/lib  -Wl,--start-group /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/embedded-file-writer.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-aix.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-base.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-generic.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-mac.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-win.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/mksnapshot.o /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_base_without_compiler.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_init.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_libbase.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_libplatform.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_nosnapshot.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_libsampler.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_compiler.a /home/develict/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_initializers.a -L/home/develict/data/homebridge_a20/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib -latomic -ldl -lrt -Wl,--end-group
/usr/bin/ld: cannot find /usr/lib/libatomic.so.1.0.0
collect2: error: ld returned 1 exit status
```

zokl, 2020-03-08:
I am using Centos 7. I found that after installing package libatomic.i686 compilation finishes without error.

Libatomic package and its 32-bit version is not a standard package of OpenWRT compilation environment.

nxhack, 2020-03-09:
Tell me your build environment.
　openwrt version (branch)

nxhack, 2020-03-09:
Node.js builds both the host side and cross builds for the target side. This error was due to the environment of host itself.

zokl, 2020-03-09:
> Node.js builds both the host side and cross builds for the target side. This error was due to the environment of host itself.

The situation is more complicated. Node.js in combination with ARM 32-bin environment needs more host packages in the host environment (libatomic (64, 32 bin), libstdc++ (64, 32 bit), etc. ). The main issue is a requirement of GCC > 6.3. Centos 7 has only GCC 4.8. 

All issues mentioned earlier could be resolved in the host environment. 

Is it possible to put information about the requirement into the description of the package?

nxhack, 2020-03-09:
I will add descriptions at next update.

https://github.com/nodejs/node/blob/v12.x/BUILDING.md#supported-toolchains

And I have already described the need for multilib packages when building 32-bit targets.

zokl, 2020-03-09:
> I will add descriptions at next update.
> 
> https://github.com/nodejs/node/blob/v12.x/BUILDING.md#supported-toolchains
> 
> And I have already described the need for multilib packages when building 32-bit targets.

Thanks a lot. I have also used this information. They do not list everything, but they are sufficient to guide them in the right direction. The Internet will then help with the rest :-).

closed 2020-08-17 by BKPepe