openwrt/packages #11534: wget: ssl and nossl variant should provide different packages

bk138, 2020-03-09:
Maintainer: @@tripolar
Environment: mips_24kc, UNIFI AP AC PRO; OpenWRT 18.06.5

Description: We have a package here that depends on wget with ssl (as was the defaut 15.05). On 18.06, that package pulls in wget-nossl because that provides `wget`. There is currently no way we know of to specifically depend on the -ssl version. Maybe additional `Provides: ` should be specified.

neheb, 2020-03-09:
ping @yousong

neheb, 2020-03-09:
This was fixed with 19.07 btw.

bk138, 2020-03-09:
> This was fixed with 19.07 btw.

Can't really find this in https://github.com/openwrt/packages/blob/openwrt-19.07/net/wget/Makefile - or do you mean https://github.com/openwrt/packages/blob/master/net/wget/Makefile ? If I'm not mistaken, the master branch let's package `wget` (the ssl variant) provide `gnu-wget` and package `wget-nossl` provide `gnu-wget,wget`.

neheb, 2020-03-09:
Hrm I see what you mean.

What does opkg install gnu-wget install?

Is GNU wget a requirement or would uclient-fetch with libustream-ssl work?

bk138, 2020-03-10:
`opkg install gnu-wget`, as `opkg install wget`, installs `wget`. The problem/challenge seems to be that packages pulled in via `Depends:`  are resolved differently, it kinda seems that `Provides:` takes precedence over the package name. But this is just from a layer 8 perspective.

Jason-H, 2020-04-04:
I ran into this today while installing and running `acme` on 19.07.2, which depends on `gnu-wget`. I also reproduced this with a clean install of 19.07.2. When running `opkg install acme`, the package `wget-nossl` is downloaded. After that, acme fails since the api is called over https.

yousong, 2020-04-06:
I opened a pull request proposing new provides name "wget-ssl".  It's in rfc stage because I'd like to see if the addition is really essential, or could we make it easier to grasp without breaking existing users' expectation.

bk138, 2020-04-06:
Just my 2 cents: I would expect ´wget´ to provide the SSL variant and ´wget-nossl´ the stripped-down special-use-case package. Is there an OpenWrt policy documented on this?

yousong, 2020-04-06:
> Just my 2 cents: I would expect ´wget´ to provide the SSL variant and ´wget-nossl´ the stripped-down special-use-case package. Is there an OpenWrt policy documented on this?

When uclient-fetch was developed as a minimal replacement for gnu wget, it was packaged to `PROVIDES` wget.

Last time we came across this issue, it's about support for `--headers` argument (https://github.com/openwrt/packages/issues/9456#issuecomment-510738475).  I still think `PROVIDES` names should try to give the intuition that the feature set will be fulfilled by all its providers.  But as far as I know, there is no documented policy on this.

bk138, 2020-04-06:
Personally, I tend to think of "Provides: wget" that I'd get the wget feature set as known from desktop distros. But of course YMMV.

jow-, 2020-04-06:
One might ask the question what the point of `wget-nossl` actually is. I would think that the niche for use-cases that require GNU wget instead of BusyBox wget but without SSL support is vanishingly small.

neheb, 2020-04-06:
Funny. I meant to write something similar but deleted before I posted.

yousong, 2020-04-07:
> One might ask the question what the point of `wget-nossl` actually is. I would think that the niche for use-cases that require GNU wget instead of BusyBox wget but without SSL support is vanishingly small.

BusyBox wget is not a contender here as we have it disabled by default in the build.  If we enable it by default we have no use for minimalism of uclient-fetch ;)

In the end it's about bytes on non-volatile storage.  I collected some stats about *compressed* size of them from downloads.openwrt.org

 - TLS library: libmbedtls(174KB) is about 795KB smaller than libopenssl(969KB)
 - Executables:
    - size of uclient-fetch is negligible
    - curl about 128KB smaller than wget with ssl, 108KB smaller than wget-nossl
    - curl as with uclient-fetch can work with almost all tls libraries while the wget build can only work with openssl

So if it's just about tls support, uclient-fetch with libmbedtls is a clear win.

If it's also about feature sets, size of curl (50KB+174KB) is likely still a better choice than wget (128KB + 969KB).  Unless gnu wget is really required, I would suggest using curl wherever possible.  In the case of acme (ping @tohojo) , wget is a better choice because it also needs openssl-util...

Now, should we keep wget-nossl, or not?  It costs infra resources but causes no harm when it comes to opkg usage.  The name is clear and I do not expect it to cause confusion when put it side to side with wget, wget-nossl, uclient-fetch etc.  In addition I see non-negligible number of downloads of wget-nossl in [stats page](http://downloads.openwrt.org/stats/awstats.downloads.openwrt.org.allextra2.html).  So I see no hurry to remove it.

| Name                                                           	| Size     	|
|----------------------------------------------------------------	|----------	|
| ca-certificates_20190110-2_all.ipk                             	| 122.7 KB 	|
| libopenssl1.1_1.1.1f-1_mips_24kc.ipk                           	| 969.4 KB 	|
| libmbedtls12_2.16.5-1_mips_24kc.ipk                            	| 174.1 KB 	|
| libustream-mbedtls20200215_2020-03-13-5e1bc342-1_mips_24kc.ipk 	| 4.8 KB   	|
| uclient-fetch_2020-02-15-af585dbd-1_mips_24kc.ipk              	| 7.3 KB   	|
| wget-nossl_1.20.3-3_mips_24kc.ipk                              	| 158.6 KB 	|
| wget_1.20.3-3_mips_24kc.ipk                                    	| 178.3 KB 	|
| curl_7.69.1-1_mips_24kc.ipk                                    	| 50.9 KB  	|
| libcurl4_7.69.1-1_mips_24kc.ipk                                	| 117.8 KB 	|

tohojo, 2020-04-07:
Yousong Zhou <notifications@github.com> writes:

> If it's also about feature sets, size of curl (50KB+174KB) is likely
> still a better choice than wget (128KB + 969KB).  Unless gnu wget is
> really required, I would suggest using curl wherever [...]

acme.sh will quite happily use curl instead of wget; so if we're doing
all this just because of the acme dependency, we could just switch that
over to curl. However, as you say acme.sh also relies on the openssl
binary, so we'll end up with a larger total binary size...

yousong, 2020-04-18:
I just tested result of #11775 with x86/64 on QEMU.  The issue should now be fixed on master branch.

I just made two backport pull requests for 18.06 and 19.07.  They both contain the same series of commits leading to those in #11775.

yousong, 2020-04-20:
Tested with an old 18.06.5 x86/64 image on QEMU, the issue should now be fixed by the aforementioned changes.

In short, a new name `wget-ssl` was introduced to allow other packages to select an implementation of wget with ssl support.  See net/acme/Makefile as an example

closed 2020-04-20 by yousong