openwrt/packages #11539: hostapd repeatedly dropping all clients

thatso, 2020-03-09:
I'm currently running OpenWrt 19.07.2 on a MR18 with ar71xx target and experiencing for some time now that hostapd out of the blue drops all clients. They can only reconnect if I manually restart the process. This started to happen without apparent reason even in previous versions and is bugging me ever since.
CPU/system/memory loads are constantly low and never hitting any resource limits.

When it happens, the system log is filled with entries like this:
`Mon Mar  9 12:12:34 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:16:dc:xx:xx:xx`
`Mon Mar  9 12:12:34 2020 daemon.info hostapd: wlan0: STA 00:16:dc:xx:xx:xx IEEE 802.11: disassociated due to inactivity`
`Mon Mar  9 12:12:35 2020 daemon.info hostapd: wlan0: STA 00:16:dc:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)`

However, all clients are active despite the messages suggesting otherwise.
I've already disabled "Disassociate On Low Acknowledgement" to no avail.

neheb, 2020-03-09:
https://bugs.openwrt.org

closed 2020-03-09 by neheb

thatso, 2020-03-09:
I've created [task 2892](https://bugs.openwrt.org/index.php?do=details&task_id=2892) if that is what you were trying to suggest.
A few words of explanation would have been really nice besides just closing this issue.

neheb, 2020-03-09:
hostapd is part of the master repository. The developers there don't care much for this one.