openwrt/packages #11547: msmtp in OpenWrt 18.06 will prevent to use sending mails from Google next year

BKPepe, 2020-03-10:
Maintainer: @nicolas-thill 
Environment: any router using msmtp 1.6.6 on OpenWrt 18.06.

Description:

In OpenWrt 18.06, there is a version of msmtp, which does not support SNI (this was added in 1.6.8) and OAUTHBEARER/XOAUTH2. It was added since version 1.8.4 and later fixed in 1.8.5.   This is a little bit problematic as the Google next year will remove "less secure app access" and it will prevent to sending emails through smtp.gmail.com

Cross reference: https://forum.turris.cz/t/update-msmtp-1-8-4-for-oauthbearer-xoauth2-support/12434

In `master` branch, there is a version 1.8.7, which should be included in `OpenWrt 19.07`, where is version 1.8.4.

neheb, 2020-03-11:
I took over as maintainer in master.

Feel free to backport it.

I wanted to port all the scripts to normal POSIX sh instead of bash but didn't get too far.

neheb, 2020-04-08:
Closing since https://github.com/openwrt/packages/pull/11773 got merged.

closed 2020-04-08 by neheb