openwrt/packages #11554: postfix: smtputf8_enable is true, but EAI support is not compiled i

ojkastl, 2020-03-11:
# Issue template (remove lines from top till here)

Maintainer: @Shulyaka
Environment: 19.07.1 running on x86/geode

Description:
Postfix warns during installation that EAI support is not compiled in

```
# opkg install postfix
Installing postfix (3.4.4-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/i386_pentium/packages/postfix_3.4.4-2_i386_pentium.ipk
Configuring postfix.
postfix: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix: warning: smtputf8_enable is true, but EAI support is not compiled in
newaliases: warning: smtputf8_enable is true, but EAI support is not compiled in
postalias: warning: smtputf8_enable is true, but EAI support is not compiled in
postmap: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix: warning: smtputf8_enable is true, but EAI support is not compiled in
postsuper: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix: warning: smtputf8_enable is true, but EAI support is not compiled in
postsuper: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix/postlog: warning: smtputf8_enable is true, but EAI support is not compiled in
postfix/postfix-script: starting the Postfix mail system
```

Can be suppressed using this stanza in main.cf, but would be nice if this could be compiled in.
```
smtputf8_enable = no
```

Shulyaka, 2020-03-11:
You can compile with smtputf8 support using configuration options. The reason it is not enabled by default is the significant increase of package size.

I agree, we need to include `smtputf8_enable = no` when the support is not compiled.

johanneskastl, 2020-03-11:
> You can compile with smtputf8 support using configuration options. The reason it is not enabled by default is the significant increase of package size.

Could this be an additional package postfix-full, then? I see many others (vim, etc.) having more than one variant, depending on the needs?