openwrt/packages #11557: SMB: Windows 10 can't access folders on OpenWRT SMB server

PavelSosin-320, 2020-03-11:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer:  (@Andy2244 
Environment: Technicolor DGA2232, version Cobalt 18.3 (old!)

Description:
Can't open a folder from Windows 10 File explorer 
Expected: folder is opened like from all my Linux-like OS clients: Android, Tyzen, etc. 
Actual:
 Windows message "
[image]

It looks like this issue is well known and solved for Ubuntu and SMB 4  [known solution](https://askubuntu.com/questions/1133578/windows-10-not-connecting-properly-with-samba-4-7-6-on-ubuntu-18-04/1141477#1141477?newreg=bdeeb006854f4c3a9d83ec6a876f3e53)
In which build of openWRT this issue is solved and can be applied to my router?
P.S. I got this router as OEM from my ISP and can't achieve developers' support.
P.S.S: I use developers' pre-release of  Windows 10
```
Formating code blocks by wrapping them with pairs of ```
```

Andy2244, 2020-03-11:
>  version Cobalt 18.3 (old!)

If this is your openwrt version than its very old and unsupported, also anything older than 19.x is only supporting samba36 not samba4.

My Samba4 package for 19.x+ works fine out of the box with windows 10, so there is no fix needed. For samba36 you might need to add those settings to your `smb.conf.template` under the `[global]` section:
```
lanman auth = yes
ntlm auth = ntlmv1-permitted
```

The reason is that windows 10 does not by default support smb1 and the older insecure authentication methods anymore. Alternatively you can enable smb1 support in windows 10 again, if LAN security is not important. https://winaero.com/blog/enable-smb1-sharig-protocol-windows-10/

PavelSosin-320, 2020-03-12:
@Andy2244, first of all, I have smb1.0 enabled as you can see below but it doesn't help. But I afraid the problem lays much deeper - we are talking about different Windows 10 versions. I'm running the 2002006 development pre-release given me by Microsoft to play and test.  In this version, many things look different. In File explorer network tab I see all my devices in my local network and their properties harvested by Windows including software version and information from the manufacturer site.  I see my Technicolor router with DNLA service which is associated with Windows Media player. I also see all my printers, scanners and, even Android phones. But I see 2 unrecognized devices with no information available. Click or Open these devices results in "remote procedure call failure". Looks like Windows now expects some capabilities response according to so common Microsoft protocols pattern.
Do you know whether OpenWRT Docker container contains SMB server?
[image]

PavelSosin-320, 2020-03-12:
Additional information
1. Technicolor DNLA service properties see below
[image]

Powershell Set-SmbClientConfiguration -EnableInsecureGuestLogons switch between 0 to 1 has no effect

Andy2244, 2020-03-12:
I still don't know what openWRT version you are actually running and what samba server version?
Also by default windows 10 will not list samba shares in the explorer/network places anymore, for this you need the wsdd2 package, so try access the share via its full UNC name aka `\\Technicolor\(sharename)` or ip `\\192.x.x.x\share`.

PavelSosin-320, 2020-03-12:
I'm running an old version of openWRT Cobalt 18.3 which was installed by my ISP on OEM Technicolor router which they supplied to me without any documentation and technical support. I have a home network with average complexity for today: WiFi router + Samsung smart TV (Tyzer Unix from Samsung) + Windows10 laptops + Android phones. I can connect to DLNA service from phones and navigate through router's FS from phones (fixed SMB 2.0 version). But I can see only DLNA services on the same router from Windows and TV. TV also doesn' t show Router's SMB as a possible source. So, I can conclude that the issue is not in SMB2.0 implementation but in how router responses on service discovery. Why DLNA is correctly discovered in my network and SMB not?
Which SMB version openWRT promises to support and how to negotiate protocol dialect?
I would like to send this question to Microsoft as a fast ring insider to get a response to the simple question: What Windows 10 exactly needs from SMB 2.0  server to discover it correctly. And also what is it SMBdirect?
Meanwhile, I'm closing this thread and will return to you when MS gives us an answer.

closed 2020-03-12 by PavelSosin-320

PavelSosin-320, 2020-04-06:
Thank you very much! This is exactly my case. I didn't find a solution in
this thread but, at least, gives some direction.