openwrt/packages #11561: Eror /libboost_graph*.so*': No such file or directory. openwrt 19.07.2 ath79

serverlinuxit, 2020-03-12:
`     cp: cannot stat '/home/clasic/passwall/openwrt-sdk-19.07.2-ath79-generic_gcc-7.5.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/boost_1_71_0/ipkg-install/lib/libboost_graph*.so*': No such file or directory
Makefile:538: recipe for target '/home/clasic/passwall/openwrt-sdk-19.07.2-ath79-generic_gcc-7.5.0_musl.Linux-x86_64/bin/packages/mips_24kc/packages/boost-graph_1.71.0-6_mips_24kc.ipk' failed
make[2]: *** [/home/clasic/passwall/openwrt-sdk-19.07.2-ath79-generic_gcc-7.5.0_musl.Linux-x86_64/bin/packages/mips_24kc/packages/boost-graph_1.71.0-6_mips_24kc.ipk] Error 1
make[2]: Leaving directory '/home/clasic/passwall/openwrt-sdk-19.07.2-ath79-generic_gcc-7.5.0_musl.Linux-x86_64/feeds/packages/libs/boost'
time: package/feeds/packages/boost/compile#0.88#0.63#1.54
package/Makefile:111: recipe for target 'package/feeds/packages/boost/compile' failed
make[1]: *** [package/feeds/packages/boost/compile] Error 2
make[1]: Leaving directory '/home/clasic/passwall/openwrt-sdk-19.07.2-ath79-generic_gcc-7.5.0_musl.Linux-x86_64'
/home/clasic/passwall/openwrt-sdk-19.07.2-ath79-generic_gcc-7.5.0_musl.Linux-x86_64/include/toplevel.mk:218: recipe for target 'package/luci-app-passwall/compile' failed
make: *** [package/luci-app-passwall/compile] Error 2     `

neheb, 2020-03-22:
Need a fuller log to see what's happening.

ClaymorePT, 2020-05-07:
@serverlinuxit Can you provide a complete build log?

closed 2020-08-17 by BKPepe