openwrt/packages #11562: libgpiod: build fails when adding libgpiod and gpio-utils

userABCD-ship-it, 2020-03-12:
Maintainer: @mhei 
Environment: (MIPs24kc, Atheros AR9331, OpenWrt SNAPSHOT, r12154-6c407fb5db)

Description:
Include both libgpiod and gpio-utils in make menuconfig and the build fails.
``
define Package/gpiod-tools/install
	$(INSTALL_DIR) $(1)/usr/bin
	**$(CP) $(PKG_INSTALL_DIR)/usr/bin/* $(1)/usr/bin/**
endef
``
openwrt/build_dir/target-mips_24kc_musl/libgpiod-1.4.1/ipkg-install/usr/bin/*': No such file or directory

The attached part of the makefile tries to do a copy command from a directory that does not exist.

mhei, 2020-03-14:
I cannot reproduce this. Can you give details? For example run scripts/diffconfig.sh and append the output here.

userABCD-ship-it, 2020-03-17:
`CONFIG_TARGET_ar71xx=y
CONFIG_TARGET_ar71xx_generic=y
CONFIG_TARGET_ar71xx_generic_Default=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_gpiod-tools=y
CONFIG_PACKAGE_libgpiod=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_libsensors=y
CONFIG_PACKAGE_libsysfs=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_lm-sensors=y
CONFIG_PACKAGE_lm-sensors-detect=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_perl=y
CONFIG_PACKAGE_perlbase-base=y
CONFIG_PACKAGE_perlbase-bytes=y
CONFIG_PACKAGE_perlbase-class=y
CONFIG_PACKAGE_perlbase-config=y
CONFIG_PACKAGE_perlbase-cwd=y
CONFIG_PACKAGE_perlbase-errno=y
CONFIG_PACKAGE_perlbase-essential=y
CONFIG_PACKAGE_perlbase-fcntl=y
CONFIG_PACKAGE_perlbase-file=y
CONFIG_PACKAGE_perlbase-filehandle=y
CONFIG_PACKAGE_perlbase-i18n=y
CONFIG_PACKAGE_perlbase-integer=y
CONFIG_PACKAGE_perlbase-io=y
CONFIG_PACKAGE_perlbase-list=y
CONFIG_PACKAGE_perlbase-locale=y
CONFIG_PACKAGE_perlbase-params=y
CONFIG_PACKAGE_perlbase-posix=y
CONFIG_PACKAGE_perlbase-re=y
CONFIG_PACKAGE_perlbase-scalar=y
CONFIG_PACKAGE_perlbase-selectsaver=y
CONFIG_PACKAGE_perlbase-socket=y
CONFIG_PACKAGE_perlbase-symbol=y
CONFIG_PACKAGE_perlbase-tie=y
CONFIG_PACKAGE_perlbase-unicore=y
CONFIG_PACKAGE_perlbase-utf8=y
CONFIG_PACKAGE_perlbase-xsloader=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_sysfsutils=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y`

`cp -fpR /home/openwrt/build_dir/target-mips_24kc_musl/libgpiod-1.4.1/ipkg-install/usr/bin/* /home/openwrt/build_dir/target-mips_24kc_musl/libgpiod-1.4.1/ipkg-mips_24kc/gpiod-tools/usr/bin/
cp: cannot stat '/home/openwrt/build_dir/target-mips_24kc_musl/libgpiod-1.4.1/ipkg-install/usr/bin/*': No such file or directory
Makefile:106: recipe for target '/home/openwrt/bin/packages/mips_24kc/packages/gpiod-tools_1.4.1-2_mips_24kc.ipk' failed
make[3]: *** [/home/openwrt/bin/packages/mips_24kc/packages/gpiod-tools_1.4.1-2_mips_24kc.ipk] Error 1`

mhei, 2020-03-19:
Thanks, but I still cannot reproduce. I used a fresh, clean checkout of OpenWrt master and packages master, create `.config `with your settings, run `make menuconfig` and then `make`. Build succeeded without problems.

userABCD-ship-it, 2020-03-19:
I did a fresh pull and build and it appeared to work. May have been working with stale data.

closed 2020-03-19 by userABCD-ship-it