openwrt/packages #11571: wman3 quic

radiy, 2020-03-14:
Maintainer: @brianjmurrell

Description: default config should be updated to provide quic protocol template.
Wasted few hours trying to undestend why google returns wrong ip for "my ip" request, if you use chrome it uses quic protocol and it`s udp on same port as https.

```
config rule 'quic'
        option sticky '1'
        option dest_port '443'
        option proto 'udp'
        option use_policy 'wan2_only'
```

brianjmurrell, 2020-03-14:
@radiy I am not the maintainer, @feckert is.

radiy, 2020-03-14:
Sorry about it, trying to do what template asked me to do - look for maintainer in makefile log.

Strykar, 2020-12-19:
@radiy Title should be `mwan3` not `wman3`.. threw me off a for minute.
@feckert Perhaps a rule hint in the (extensive) hint list on top of `cgi-bin/luci/admin/network/mwan/rule` would suffice?

closed 2020-12-21 by feckert