openwrt/packages #11584: luci-proto-openconnect: user cert and key file naming error

DasTestament, 2020-03-17:
Maintainer: @nmav @jow- 
Environment: ramips, xiaomi r3g, 19.07.2

Description:
Steps to reproduce:
1. Create OpenConnect interface (if0)
2. Fill **User certificate** or **User key** or **CA certificate** field and save
3. Save and Apply
4. Go to **/etc/openconnect** and observe that related PEM file is named without **vpn-** before iface name:
```
ca-if0.pem
user-cert-if0.pem
user-key-if0.pem
```
but openconnect.sh requires PEM to be named with **vpn-** part like this:
```
ca-vpn-if0.pem
user-cert-vpn-if0.pem
user-key-vpn-if0.pem
```

nmav, 2020-03-20:
Hi,
 Would you like to send a pull request to fix this?

DasTestament, 2020-03-23:
@nmav Hello,

Sure, why not. Give me some time, I'm a bit busy atm.

DasTestament, 2020-03-26:
@nmav I've provided a fix for this case in PR #3786

nmav, 2020-04-08:
Addressed in luci.

closed 2020-04-08 by nmav

sunnyqeen, 2020-04-21:
@nmav @jow- 
Could you please merge it to 19.07 branch?
https://github.com/openwrt/luci/commit/adf0fb1879ffb6bd04984d257484fad0fba1d6a3
Thanks!