openwrt/packages #11588: prometheus-statsd-exporter: does not compile since resent golang update

feckert, 2020-03-18:
Maintainer: @dhewg 
Environment: x86_64, OpenWrt master

Description:

Recently, prometheus-stasd-exporter no longer builds on the x86_64 target. I think it has to do with the following changes https://github.com/openwrt/packages/commit/074b417c674109795a1bb88759ae4877f55fc679 @jefferyto any hint from your side what could be the problem?

```
touch /home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.built
touch /home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.built_check
( cd /home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.go_work/build ; export               GOOS=linux GOARCH=amd64 GO386= GOARM= GOMIPS= GOMIPS64= CGO_ENABLED=1 CC=x86_64-openwrt-linux-musl-gcc CXX=x86_64-openwrt-linux-musl-g++ CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1=statsd_exporter-0.8.1 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1=statsd_exporter-0.8.1 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/target-x86_64_musl/lib -L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -fpic -specs=/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/include/hardened-ld-pie.specs -znow -zrelro"         GOPATH=/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.go_work/build GOCACHE=/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.go_work/cache GOENV=off ; echo "Finding targets" ; targets=$(go list github.com/prometheus/statsd_exporter/...) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v  $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case amd64 in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="" ;; mips64|mips64le) installsuffix="" ;; esac ; ldflags="-linkmode external -extldflags '-L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/target-x86_64_musl/lib -L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -fpic -specs=/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/include/hardened-ld-pie.specs -Wl,-z,now -Wl,-z,relro'" ; pkg_gcflags="" ; pkg_ldflags="" ; for def in ; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install -buildmode pie ${installsuffix:+-installsuffix $installsuffix} -trimpath -ldflags "all=$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags $ldflags"}  $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1/.go_work/build/bin -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -modcache ; echo ; fi ; exit $retval ; )
Finding targets
go: downloading github.com/prometheus/common v0.0.0-20170731114204-61f87aac8082
go: downloading github.com/prometheus/client_golang v0.8.0
go: downloading gopkg.in/yaml.v2 v2.2.1
go: downloading github.com/howeyc/fsnotify v0.0.0-20151003194602-f0c08ee9c607
go: downloading gopkg.in/alecthomas/kingpin.v2 v2.2.5
go: downloading github.com/golang/protobuf v1.2.0
go: downloading github.com/beorn7/perks v0.0.0-20160804104726-4c0e84591b9a
go: downloading github.com/prometheus/client_model v0.0.0-20170216185247-6f3806018612
go: downloading github.com/prometheus/procfs v0.0.0-20170703101242-e645f4e5aaa8
go: downloading github.com/alecthomas/units v0.0.0-20151022065526-2efee857e7cf
go: downloading github.com/alecthomas/template v0.0.0-20160405071501-a0175ee3bccc
go: downloading github.com/sirupsen/logrus v1.0.3
go: downloading github.com/matttproud/golang_protobuf_extensions v1.0.0
go: downloading golang.org/x/sys v0.0.0-20180909124046-d0be0721c37e
go: downloading golang.org/x/crypto v0.0.0-20170825220121-81e90905daef

Building targets
go: inconsistent vendoring in /home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/build_dir/target-x86_64_musl/statsd_exporter-0.8.1:
        github.com/alecthomas/assert@v0.0.0-20170929043011-405dbfeb8e38: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/alecthomas/colour@v0.0.0-20160524082231-60882d9e2721: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/alecthomas/repr@v0.0.0-20181024024818-d37bc2a10ba1: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/alecthomas/template@v0.0.0-20160405071501-a0175ee3bccc: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/alecthomas/units@v0.0.0-20151022065526-2efee857e7cf: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/beorn7/perks@v0.0.0-20160804104726-4c0e84591b9a: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/davecgh/go-spew@v1.1.1: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/howeyc/fsnotify@v0.0.0-20151003194602-f0c08ee9c607: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/kr/pretty@v0.1.0: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/mattn/go-isatty@v0.0.4: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/matttproud/golang_protobuf_extensions@v1.0.0: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/onsi/ginkgo@v1.7.0: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/onsi/gomega@v1.4.3: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/pmezard/go-difflib@v1.0.0: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
        github.com/prometheus/client_golang@v0.8.0: is explicitly required in go.mod, but not marked as explicit in vendor/modules.txt
[... cut ...]
time: package/feeds/packages/prometheus-statsd-exporter/compile#1.74#0.48#1.65
make[1]: *** [package/Makefile:113: package/feeds/packages/prometheus-statsd-exporter/compile] Error 2
make[1]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt'
make: *** [/home/feckert/workspace/openwrt/LDM-master-64-prepare/build/openwrt/include/toplevel.mk:227: package/feeds/packages/prometheus-statsd-exporter/compile] Error n 'go mod vendor' to sync, or use -mod=mod or -mod=readonly to ignore the vendor directory
```

jefferyto, 2020-03-18:
I think Go 1.14 has gotten more strict about the `vendors` directory, but the version of statsd_exporter packaged here ([0.8.1](https://github.com/prometheus/statsd_exporter/releases/tag/v0.8.1)) is from December 2018.

I would prefer updating to a current version (and fixing any issues that may appear) rather than trying to get this version working.

dhewg, 2020-03-18:
Indeed, a recent version might just fix this. I'm swamped with work atm, wanna give it a try @feckert?

feckert, 2020-03-23:
fixd by #11593

closed 2020-05-05 by feckert